project-chip/connectedhomeip #13602: [esp32] Issue with `menuconfig`

turon, 2022-01-14:
#### Problem
Trying to build the all-clusters-app for ESP fails with `Failed to resolve component 'openthread'` on Mac:

```
$ idf.py menuconfig

Executing action: menuconfig
Running cmake in directory /Users/mturon/src/nest/connectedhomeip/turon/examples/all-clusters-app/esp32/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /Users/mturon/src/nest/connectedhomeip/turon/examples/all-clusters-app/esp32"...
-- Found Git: /usr/local/git/current/bin/git (found version "2.34.1.703.g22d0c6ccf7-goog") 
-- IDF_TARGET not set, using default target: esp32
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /Users/mturon/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Users/mturon/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Users/mturon/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project version: test_event_4_2021_07_06-2338-ga11413e3d
-- Building ESP-IDF components for target esp32
CMake Error at /Users/mturon/tools/esp-idf/tools/cmake/build.cmake:186 (message):
  Failed to resolve component 'openthread'.
Call Stack (most recent call first):
  /Users/mturon/tools/esp-idf/tools/cmake/build.cmake:217 (__build_resolve_and_add_req)
  /Users/mturon/tools/esp-idf/tools/cmake/build.cmake:426 (__build_expand_requirements)
  /Users/mturon/tools/esp-idf/tools/cmake/project.cmake:396 (idf_build_process)
  CMakeLists.txt:34 (project)

-- Configuring incomplete, errors occurred!
See also "/Users/mturon/src/nest/connectedhomeip/turon/examples/all-clusters-app/esp32/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1
```

dhrishi, 2022-01-16:
@turon Please check the ESP-IDF version you are using. CHIP now requires IDF v4.4-beta1 pre release. See [here](https://github.com/project-chip/connectedhomeip/tree/master/examples/all-clusters-app/esp32#building-the-example-application). Component `openthread` is present in that version.

turon, 2022-02-08:
I updated to release/v4.4 and get this:

```
$ idf.py set-target esp32
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Directory '/Users/mturon/src/nest/connectedhomeip/turon/examples/all-clusters-app/esp32/build' doesn't seem to be a CMake build directory. Refusing to automatically delete files in this directory. Delete the directory manually to 'clean' it.
mturon-macbookpro1:esp32 mturon$ idf.py menuconfig
Executing action: menuconfig
Running cmake in directory /Users/mturon/src/nest/connectedhomeip/turon/examples/all-clusters-app/esp32/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /Users/mturon/src/nest/connectedhomeip/turon/examples/all-clusters-app/esp32"...
-- IDF_TARGET not set, using default target: esp32
-- Found Git: /usr/local/git/current/bin/git (found version "2.35.0.263.gb82422642f-goog") 
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /Users/mturon/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Users/mturon/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Users/mturon/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
CMake Error at /Users/mturon/tools/esp-idf/tools/cmake/build.cmake:200 (message):
  Failed to resolve component 'chip'.
Call Stack (most recent call first):
  /Users/mturon/tools/esp-idf/tools/cmake/build.cmake:231 (__build_resolve_and_add_req)
  /Users/mturon/tools/esp-idf/tools/cmake/build.cmake:512 (__build_expand_requirements)
  /Users/mturon/tools/esp-idf/tools/cmake/project.cmake:378 (idf_build_process)
  CMakeLists.txt:34 (project)

-- Configuring incomplete, errors occurred!
See also "/Users/mturon/src/nest/connectedhomeip/turon/examples/all-clusters-app/esp32/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1
```

turon, 2022-02-08:
When I use `v4.4-beta1` I get a failure during `. ./export.sh` (on Mac):

```
esp-idf mturon$ . ./export.sh
Detecting the Python interpreter
Checking "python" ...
Checking "python3" ...
Python 3.9.7
"python3" has been detected
Adding ESP-IDF tools to PATH...
Using Python interpreter in /Users/mturon/.espressif/python_env/idf4.4_py3.9_env/bin/python
Checking if Python packages are up to date...
Python requirements from /Users/mturon/tools/esp-idf/requirements.txt are satisfied.
Added the following directories to PATH:
  /Users/mturon/tools/esp-idf/components/esptool_py/esptool
  /Users/mturon/tools/esp-idf/components/espcoredump
  /Users/mturon/tools/esp-idf/components/partition_table
  /Users/mturon/tools/esp-idf/components/app_update
  /Users/mturon/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin
  /Users/mturon/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-8.4.0/xtensa-esp32s2-elf/bin
  /Users/mturon/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-8.4.0/xtensa-esp32s3-elf/bin
  /Users/mturon/.espressif/tools/riscv32-esp-elf/esp-2021r2-8.4.0/riscv32-esp-elf/bin
  /Users/mturon/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin
  /Users/mturon/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin
  /Users/mturon/.espressif/tools/openocd-esp32/v0.10.0-esp32-20210902/openocd-esp32/bin
  /Users/mturon/.espressif/python_env/idf4.4_py3.9_env/bin
  /Users/mturon/tools/esp-idf/tools
Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

Traceback (most recent call last):
  File "/Users/mturon/tools/esp-idf/tools/idf.py", line 812, in <module>
    main()
  File "/Users/mturon/tools/esp-idf/tools/idf.py", line 730, in main
    cli(sys.argv[1:], prog_name=PROG, complete_var='_IDF.PY_COMPLETE')
  File "/Users/mturon/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages/click/core.py", line 1128, in __call__
    return self.main(*args, **kwargs)
  File "/Users/mturon/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages/click/core.py", line 1048, in main
    self._main_shell_completion(extra, prog_name, complete_var)
  File "/Users/mturon/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages/click/core.py", line 1123, in _main_shell_completion
    rv = shell_complete(self, ctx_args, prog_name, complete_var, instruction)
  File "/Users/mturon/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages/click/shell_completion.py", line 45, in shell_complete
    echo(comp.source())
  File "/Users/mturon/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages/click/shell_completion.py", line 326, in source
    self._check_version()
  File "/Users/mturon/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages/click/shell_completion.py", line 314, in _check_version
    raise RuntimeError(
RuntimeError: Shell completion is not supported for Bash versions older than 4.4.
```

turon, 2022-02-08:
On Linux, `v4.4-beta1` completes the `. ./export.sh`, but the menuconfig command still fails:

```
$ idf.py menuconfig
Executing action: menuconfig
Running cmake in directory /home/mturon/src/nest/connectedhomeip/turon/examples/all-clusters-app/esp32/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /home/mturon/src/nest/connectedhomeip/turon/examples/all-clusters-app/esp32"...
-- IDF_TARGET not set, using default target: esp32
-- Found Git: /usr/bin/git (found version "2.35.0.263.gb82422642f-goog") 
CMake Warning at /home/mturon/tools/esp-idf/tools/cmake/git_submodules.cmake:52 (message):
  Git submodule components/bt/controller/lib_esp32 is out of date.  Run 'git
  submodule update --init --recursive' to fix.
Call Stack (most recent call first):
  /home/mturon/tools/esp-idf/tools/cmake/build.cmake:77 (git_submodule_check)
  /home/mturon/tools/esp-idf/tools/cmake/build.cmake:174 (__build_get_idf_git_revision)
  /home/mturon/tools/esp-idf/tools/cmake/idf.cmake:50 (__build_init)
  /home/mturon/tools/esp-idf/tools/cmake/project.cmake:12 (include)
  CMakeLists.txt:20 (include)

CMake Warning at /home/mturon/tools/esp-idf/tools/cmake/git_submodules.cmake:52 (message):
  Git submodule components/bt/controller/lib_esp32c3_family is out of date.
  Run 'git submodule update --init --recursive' to fix.
Call Stack (most recent call first):
  /home/mturon/tools/esp-idf/tools/cmake/build.cmake:77 (git_submodule_check)
  /home/mturon/tools/esp-idf/tools/cmake/build.cmake:174 (__build_get_idf_git_revision)
  /home/mturon/tools/esp-idf/tools/cmake/idf.cmake:50 (__build_init)
  /home/mturon/tools/esp-idf/tools/cmake/project.cmake:12 (include)
  CMakeLists.txt:20 (include)

CMake Warning at /home/mturon/tools/esp-idf/tools/cmake/git_submodules.cmake:52 (message):
  Git submodule components/bt/host/nimble/nimble is out of date.  Run 'git
  submodule update --init --recursive' to fix.
Call Stack (most recent call first):
  /home/mturon/tools/esp-idf/tools/cmake/build.cmake:77 (git_submodule_check)
  /home/mturon/tools/esp-idf/tools/cmake/build.cmake:174 (__build_get_idf_git_revision)
  /home/mturon/tools/esp-idf/tools/cmake/idf.cmake:50 (__build_init)
  /home/mturon/tools/esp-idf/tools/cmake/project.cmake:12 (include)
  CMakeLists.txt:20 (include)

CMake Warning at /home/mturon/tools/esp-idf/tools/cmake/git_submodules.cmake:52 (message):
  Git submodule components/esp_phy/lib is out of date.  Run 'git submodule
  update --init --recursive' to fix.
Call Stack (most recent call first):
  /home/mturon/tools/esp-idf/tools/cmake/build.cmake:77 (git_submodule_check)
  /home/mturon/tools/esp-idf/tools/cmake/build.cmake:174 (__build_get_idf_git_revision)
  /home/mturon/tools/esp-idf/tools/cmake/idf.cmake:50 (__build_init)
  /home/mturon/tools/esp-idf/tools/cmake/project.cmake:12 (include)
  CMakeLists.txt:20 (include)

CMake Warning at /home/mturon/tools/esp-idf/tools/cmake/git_submodules.cmake:52 (message):
  Git submodule components/esp_wifi/lib is out of date.  Run 'git submodule
  update --init --recursive' to fix.
Call Stack (most recent call first):
  /home/mturon/tools/esp-idf/tools/cmake/build.cmake:77 (git_submodule_check)
  /home/mturon/tools/esp-idf/tools/cmake/build.cmake:174 (__build_get_idf_git_revision)
  /home/mturon/tools/esp-idf/tools/cmake/idf.cmake:50 (__build_init)
  /home/mturon/tools/esp-idf/tools/cmake/project.cmake:12 (include)
  CMakeLists.txt:20 (include)

CMake Warning at /home/mturon/tools/esp-idf/tools/cmake/git_submodules.cmake:52 (message):
  Git submodule components/expat/expat is out of date.  Run 'git submodule
  update --init --recursive' to fix.
Call Stack (most recent call first):
  /home/mturon/tools/esp-idf/tools/cmake/build.cmake:77 (git_submodule_check)
  /home/mturon/tools/esp-idf/tools/cmake/build.cmake:174 (__build_get_idf_git_revision)
  /home/mturon/tools/esp-idf/tools/cmake/idf.cmake:50 (__build_init)
  /home/mturon/tools/esp-idf/tools/cmake/project.cmake:12 (include)
  CMakeLists.txt:20 (include)

CMake Warning at /home/mturon/tools/esp-idf/tools/cmake/git_submodules.cmake:52 (message):
[... 40 lines cut ...]
-- Check for working CXX compiler: /home/mturon/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
-- Found Python3: /home/mturon/src/nest/connectedhomeip/master/.environment/pigweed-venv/bin/python3.9 (found version "3.9.9") found components: Interpreter 
-- Project sdkconfig file /home/mturon/src/nest/connectedhomeip/turon/examples/all-clusters-app/esp32/sdkconfig
Traceback (most recent call last):
  File "/home/mturon/tools/esp-idf/tools/kconfig_new/confgen.py", line 33, in <module>
    import gen_kconfig_doc
  File "/home/mturon/tools/esp-idf/tools/kconfig_new/gen_kconfig_doc.py", line 27, in <module>
    import kconfiglib
ImportError: bad magic number in 'kconfiglib': b'\x03\xf3\r\n'
CMake Error at /home/mturon/tools/esp-idf/tools/cmake/kconfig.cmake:266 (message):
  Failed to run confgen.py
  (/home/mturon/src/nest/connectedhomeip/master/.environment/pigweed-venv/bin/python;/home/mturon/tools/esp-idf/tools/kconfig_new/confgen.py;--kconfig;/home/mturon/tools/esp-idf/Kconfig;--sdkconfig-rename;/home/mturon/tools/esp-idf/sdkconfig.rename;--config;/home/mturon/src/nest/connectedhomeip/turon/examples/all-clusters-app/esp32/sdkconfig;--defaults;/home/mturon/s [... cut ...] ip/turon/examples/all-clusters-app/esp32/build/config.env).
  Error 1
Call Stack (most recent call first):
  /home/mturon/tools/esp-idf/tools/cmake/build.cmake:528 (__kconfig_generate_config)
  /home/mturon/tools/esp-idf/tools/cmake/project.cmake:378 (idf_build_process)
  CMakeLists.txt:34 (project)

-- Configuring incomplete, errors occurred!
See also "/home/mturon/src/nest/connectedhomeip/turon/examples/all-clusters-app/esp32/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1
```

turon, 2022-02-08:
Looks like `git clean -fdx` in `tools/esp-idf` directory fixes things.  I can update the docs.

PR #14912 (2022-03-21): [doc] Updates docs for cleanly upgrading to release/v4.4 ESP-IDF SDK.
#### Problem
Fix #13602

#### Change overview
Update docs with how to cleanly upgrade an existing ESP-IDF to the latest version.
Update tag to use release/v4.4.

#### Testing
Doc only change.  CI.