project-chip/connectedhomeip #13604: Add support for software version handling in OTA Requestor

carol-apple, 2022-01-14:
#### Problem
OTA Requestor needs to do various things with the software version it is updating to. A few examples:
- 1) During `QueryImageResponse`, if the software version supplied is the same as current version, ignore and do not proceed - refer to `11.20.3.2. Querying the OTA Provider`
- 2) After a software update has been applied successfully, the software version should be updated in the OTA Requestor core as well as the Basic clusters `SoftwareVersion` attribute (the OTA Requestor core should keep this value to support (1) above)

 #### Proposed Solution
Add the following to OTARequestorInterface:
- States for the current version vs target version (the new version that provider wants to supply)
- Logic to determine when to ignore an image when `QueryImageResponse` is received
- New API in OTARequestorInterface to update the software version when necessary (like after a successful apply)
- Maybe in the new API introduced above, it should call `Basic::Attributes::SoftwareVersion::Set`

The new API should be called by various implementations of OTA after an Apply has succeeded. For instance, in Linux, the `OTAImageProcessorImpl::Apply` is a no op and should call the new API directly from there with the software version it obtained maybe from the image header. For other platforms, this new API should be called after restarting the app.

dhrishi, 2022-01-20:
@jadhavrohit924 PTAL

dhrishi, 2022-01-27:
i. has been addressed and merged in #13742 
ii. updating Basic cluster attributes now no longer belongs in OTA logic