project-chip/connectedhomeip #13617: chip-device-ctrl "Segmentation fault" on "resolve" command

retfie, 2022-01-15:
#### Problem
After clean clone connectedhomeip and build python controller "chip-device-ctrl", successful commissioning over BLE thread based example (lighting-app on nrfconnect), and close ble session, attempting to resolve node ID causes several message "Re-trying resolve" and finally Segmentation fault.

 * expected behavior
Successful to update address of the node in the device controller by given node ID.

 * actual behavior
 ```
chip-device-ctrl > resolve 1234
[1642284586.188591][14294:14304] CHIP:DL: Re-trying resolve
[1642284591.190812][14294:14304] CHIP:DL: Re-trying resolve
[1642284596.195890][14294:14304] CHIP:DL: Re-trying resolve
[1642284601.198747][14294:14304] CHIP:DL: Avahi resolve failed
[1642284601.198781][14294:14304] CHIP:CTL: Error resolving node id: ../../src/platform/Linux/DnssdImpl.cpp:714: CHIP Error 0x000000AC: Internal error
[1642284601.198791][14294:14304] CHIP:CTL: Error resolving node id: ../../src/platform/Linux/DnssdImpl.cpp:714: CHIP Error 0x000000AC: Internal error
Failed to update node address: 172
[1642284601.198929][14294:14304] CHIP:CTL: Error resolving node id: ../../src/lib/dnssd/Discovery_ImplPlatform.cpp:90: CHIP Error 0x000000A6: Unknown resource ID
[1642284601.198943][14294:14304] CHIP:CTL: Error resolving node id: ../../src/lib/dnssd/Discovery_ImplPlatform.cpp:90: CHIP Error 0x000000A6: Unknown resource ID
Failed to update node address: 166
chip-device-ctrl > Segmentation fault
```

 * steps to reproduce
clone latest connectedhopmeip project, bootstrap, activate and build python chip-device-controller
try to connect and commission thread based example, close ble session, and try to resolve node ID. 
 
 * system configuration
Tried on two hosts (Ubuntu 20.04.3 LTS, Slackware 14.2+)

On node:
```
D: 1107 [DIS]DNS-SD StartServer mode=0
I: 1113 [DIS]Failed to find a valid admin pairing. Node ID unknown
I: 1119 [DIS]Start dns-sd server - no current nodeId
D: 1124 [DL]Using Thread extended MAC for hostname.
I: 1132 [DIS]Advertise commission parameter vendorID=9050 productID=20044 discriminator=3840/15
I: 1141 [DL]advertising srp service: A91DD0A8405EF0BC._matterc._udp
D: 1147 [DIS]Scheduling Discovery timeout in secs=900
I: 1188 [DL]NFC Tag emulation started
I: 2086 [DL]OnSrpClientNotification: Last requested operation completed successfully
```

On host
```
 avahi-browse -a
+   eth8 IPv4 A91DD0A8405EF0BC                              _matterc._udp        local
+   eth8 IPv6 A91DD0A8405EF0BC                              _matterc._udp        local
```

hazarmutgan, 2022-01-22:
I had the exact same issue with the ligthing-app on efr32. I just noticed today that python controller docs have been updated. https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/python_chip_controller_building.md

So I have pulled the latest main branch and followed the guide and it worked.

One thing to check, if you don't have a native ipv6 you need to add static ipv6 and the routes as mentioned here https://groups.google.com/g/openthread-users/c/jBYJ6Bn47jE/m/AzZqRzuQCgAJcd

retfie, 2022-01-23:
@hazarmutgan  thanks for the update.  
I have good communication between matter devices into thread network and all other devices over ipv6 into my local network. i.e. I can ping matter devices from any host from my local network, and vice versa  (I'm using radvd to provide global ipv6 RA prefixes.)

After sync and build chip-device-ctrl I made some progress. By commissioning new device with lighting-app on nrf52840dk_nrf52840, I noticed that now ble sessions closed automatically, and "resolve" command succeed by providing correct IP address of thread device. And after this point I was able to successfully control device over thread, although I have to invoke "close-session <node_id>" every time if device is restarted, otherwise controller can't communicate with device. Unfortunately, this doesn't help to restore communication with device after controller is stopped and then started again. On controller exit, following crash is seen: 

```
 chip-device-ctrl > exit
[1642969871.451589][2643450:2643450] CHIP:BLE: BleConnectionDelegate::CancelConnection is not implemented.
[1642969871.451770][2643450:2643450] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-yLIszu)
[1642969871.452056][2643450:2643450] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1642969871.452107][2643450:2643450] CHIP:DL: NVS set: chip-counters/total-operational-hours = 2 (0x2)
[1642969871.452123][2643450:2643450] CHIP:DL: Inet Layer shutdown
[1642969871.452136][2643450:2643450] CHIP:DL: BLE shutdown
[1642969871.452886][2643450:2643450] CHIP:DL: System Layer shutdown
free(): invalid pointer
Aborted (core dumped)
```
After this, device controller, losses ability to resolve or communicate with device with same symptoms as described in the begging of this issue:

```
chip-device-ctrl > resolve 1234
[1642970107.427127][2735371:2735378] CHIP:DL: Re-trying resolve
[1642970112.431373][2735371:2735378] CHIP:DL: Re-trying resolve
[1642970117.434571][2735371:2735378] CHIP:DL: Re-trying resolve
[1642970122.439230][2735371:2735378] CHIP:DL: Avahi resolve failed
[1642970122.439310][2735371:2735378] CHIP:CTL: Error resolving node id: ../../src/platform/Linux/DnssdImpl.cpp:714: CHIP Error 0x000000AC: Internal error
[1642970122.439340][2735371:2735378] CHIP:CTL: Error resolving node id: ../../src/platform/Linux/DnssdImpl.cpp:714: CHIP Error 0x000000AC: Internal error
Failed to update node address: 172
[1642970122.439580][2735371:2735378] CHIP:CTL: Error resolving node id: ../../src/lib/dnssd/Discovery_ImplPlatform.cpp:90: CHIP Error 0x000000A6: Unknown resource ID
[1642970122.439613][2735371:2735378] CHIP:CTL: Error resolving node id: ../../src/lib/dnssd/Discovery_ImplPlatform.cpp:90: CHIP Error 0x000000A6: Unknown resource ID
Failed to update node address: 166
Segmentation fault (core dumped)
```

Resetting device to factory defaults and new provisioning, restores ability to communicate, but only until device controller is exited or crashed. This looks pretty similar to the symptoms described in #12987.

retfie, 2022-01-28:
Update: 
TLD'R: chip-tools was able to successful resolve IP of device when correct fabricid is provided, but when trying to control it, uses wrong fabricid and commands fails with timeout.

After provisioning device with chip-device-ctrl, I was able to do successful resolve and control device over Matter (until it exits or chashed). Also fabricid is extracted to use it into chip-tool examples bellow.

```
chip-device-ctrl > resolve 1234
[1643310072.463111][2907638:2907644] CHIP:DL: Avahi resolve found
[1643310072.463172][2907638:2907644] CHIP:DIS: Node ID resolved for 0x00000000000004D2
[1643310072.463188][2907638:2907644] CHIP:DIS:     Addr 0: [fdee:b819:1dad:bc0f:791d:ce18:351b:a3eb]:5540
[1643310072.463200][2907638:2907644] CHIP:CTL: OperationalDiscoveryComplete for device ID 0x00000000000004D2
Secure Session to Device Established
[1643310072.463498][2907638:2907644] CHIP:CTL: Address resolved for node: 0x00000000000004D2
Node address has been updated
```
```
chip-device-ctrl > get-fabricid
Get fabric ID complete
Raw Fabric ID: 0x0000000000000000 (0)
Compressed Fabric ID: 0xeb81bb816058cc2f (16970051035250347055)
```

```
chip-device-ctrl > zcl OnOff Toggle 1234 1 1
[1643310303.504219][2907638:2907644] CHIP:IN: Prepared secure message 0x157e508 to 0x00000000000004D2 (1)  of type 0x8 and protocolId (0, 1) on exchange 57466i with MessageCounter:13875787.
[1643310303.504292][2907638:2907644] CHIP:IN: Sending encrypted msg 0x157e508 with MessageCounter:13875787 to 0x00000000000004D2 (1) at monotonic time: 219544548 msec
[1643310303.644412][2907638:2907644] CHIP:EM: Received message of type 0x9 with protocolId (0, 1) and MessageCounter:1812962 on exchange 57466i
[1643310303.644685][2907638:2907644] CHIP:DMG: Received Command Response Status for Endpoint=1 Cluster=0x0000_0006 Command=0x0000_0002 Status=0x0
[1643310303.645329][2907638:2907644] CHIP:IN: Prepared secure message 0x7f9585ef2190 to 0x00000000000004D2 (1)  of type 0x10 and protocolId (0, 0) on exchange 57466i with MessageCounter:13875788.
[1643310303.645392][2907638:2907644] CHIP:IN: Sending encrypted msg 0x7f9585ef2190 with MessageCounter:13875788 to 0x00000000000004D2 (1) at monotonic time: 219544690 msec
CommandResponse None
Success, no status code is attached with response.
```

Then tried chip-tool:
```
./out/host/chip-tool discover resolve 1234 0xeb81bb816058cc2f

[1643367655.489980][3041667:3041667] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-zDYiaM)
[1643367655.490147][3041667:3041667] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1643367655.490162][3041667:3041667] CHIP:DL: NVS set: chip-counters/reboot-count = 27 (0x1B)
[1643367655.490372][3041667:3041667] CHIP:DL: Got Ethernet interface: enp3s0
[1643367655.490505][3041667:3041667] CHIP:DL: Found the primary Ethernet interface:enp3s0
[1643367655.490638][3041667:3041667] CHIP:DL: Got WiFi interface: wlp5s0
[1643367655.491375][3041667:3041667] CHIP:DL: Found the primary WiFi interface:wlp5s0
[1643367655.491396][3041667:3041667] CHIP:IN: UDP::Init bind&listen port=5542
[1643367655.491417][3041667:3041667] CHIP:IN: UDP::Init bound to port=5542
[1643367655.491423][3041667:3041667] CHIP:IN: UDP::Init bind&listen port=5542
[1643367655.491443][3041667:3041667] CHIP:IN: UDP::Init bound to port=5542
[1643367655.491450][3041667:3041667] CHIP:IN: TransportMgr initialized
[1643367655.491465][3041667:3041667] CHIP:DIS: Init fabric pairing table with server storage
[1643367655.491555][3041667:3041667] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-0axUhL)
[1643367655.493181][3041667:3041667] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1643367655.493251][3041667:3041667] CHIP:DL: NVS set: chip-counters/GlobalMCTR = 27000 (0x6978)
[1643367655.493333][3041667:3041667] CHIP:ZCL: Using ZAP configuration...
[1643367655.494358][3041667:3041667] CHIP:DL: MDNS failed to join multicast group on enp3s0 for address type IPv4: ../../src/inet/UDPEndPointImplSockets.cpp:772: Inet Error 0x00000110: Address not found
[1643367655.496268][3041667:3041667] CHIP:CTL: System State Initialized...
[1643367655.499825][3041667:3041667] CHIP:CTL: Generating NOC
[1643367655.500188][3041667:3041667] CHIP:CTL: Generating ICAC
[1643367655.501108][3041667:3041667] CHIP:DIS: Verifying the received credentials
[1643367655.502918][3041667:3041667] CHIP:IN: Generating compressed fabric ID using uncompressed fabric ID 0x0000000000000001 and root pubkey
[1643367655.502960][3041667:3041667] CHIP:IN: 0x04, 0xff, 0x76, 0x38, 0x44, 0x89, 0x51, 0xdb,
[1643367655.502976][3041667:3041667] CHIP:IN: 0x42, 0x22, 0xea, 0xbe, 0xf8, 0xa9, 0xb5, 0x01,
[1643367655.502989][3041667:3041667] CHIP:IN: 0xee, 0xd4, 0x92, 0xb8, 0x29, 0x07, 0x0c, 0x84,
[1643367655.503003][3041667:3041667] CHIP:IN: 0x23, 0xe9, 0xa8, 0x9a, 0xae, 0x1f, 0x4c, 0xe1,
[1643367655.503021][3041667:3041667] CHIP:IN: 0x9c, 0x52, 0x19, 0x51, 0x3c, 0x98, 0x50, 0x43,
[1643367655.503047][3041667:3041667] CHIP:IN: 0x60, 0xce, 0xdf, 0x25, 0xae, 0x81, 0x00, 0x25,
[1643367655.503070][3041667:3041667] CHIP:IN: 0xef, 0x3a, 0x63, 0x02, 0x43, 0x0b, 0xbd, 0x95,
[1643367655.503093][3041667:3041667] CHIP:IN: 0x3a, 0xa7, 0x90, 0xc1, 0xdf, 0x85, 0x1b, 0xb5,
[1643367655.503111][3041667:3041667] CHIP:IN: 0x73,
[... 169 lines cut ...]

tcpdump for **chip-tool resolve** (working case)
```
./out/host/chip-tool discover resolve 1234 0xeb81bb816058cc2f

[... cut ...]
13:01:26.363759 IP6 fe80::f00b:a6be:1d22:d78b.59094 > ff02::fb.5353: 0 ANY (QU)? 84E610A7A226A94B-00000000000004D2._matter._tcp.local. (70)
13:01:53.524391 IP6 fe80::12cd:b6ff:fe02:8c8b.5353 > ff02::fb.5353: 0 [2q] AAAA (QM)? 1C03164PAA00620.local. A (QM)? 7A543D4A220FA535.local. (62)
13:01:53.529255 IP6 fe80::6889:354a:edaa:7345.5353 > ff02::fb.5353: 0*- [0q] 0/0/1 (54)
13:01:53.836169 IP6 fe80::ba27:ebff:fe3a:dd65.5353 > ff02::fb.5353: 0*- [0q] 0/0/1 (54)
13:02:00.905526 IP6 fe80::12cd:b6ff:fe02:8c8b.5353 > ff02::fb.5353: 0 AAAA (QM)? f2ed13db-cd9d-0ce7-1c11-e2e7d1d9de03.local. (60)
```

retfie, 2022-01-28:
Final update: **Success**

by using entirely chip-tool commands (commission and onoff), I was able to control device even after node device reboot
```
./out/host/chip-tool pairing ble-thread node-id operationalDataset setup-pin-code discriminator
./out/host/chip-tool onoff toggle node_id endpoint-id
```

erjiaqing, 2022-02-11:
After checking the code and testing locally, the actual issue is:
1. Python API (build with chip_mdns="platform") will crash when failed to resolve the node id, seems it will be built with minimal mdns in recent builds. 
2. Python API cannot persist credentials between restart (this is fixed in #13715 , please give chip-repl a try if possible)

mrjerryjohns, 2022-03-03:
Is this still an issue? @retfie please confirm.

closed 2022-12-21 by stale[bot]