project-chip/connectedhomeip #13645: Attributes and commands of "WiFi Network Diagnostics" cluster are failing

manjunathambiger23, 2022-01-18:
Summary:
               All attributes and commands of WiFi Network Diagnostics cluster showing Failure response when made a run.
Attributes:
BSSID, SecurityType, WiFiVersion, ChannelNumber, RSSI, BeaconLostCount, BeaconRxCount, PacketMulticastRxCount, PacketMulticastTxCount, PacketUnicastRxCount, PacketUnicastTxCount, CurrentMaxRate, OverrunCount
Command:
ResetCounts

Spec screenshot:
[image]
[image]
[image]

Execution screenshot and Log files:
[image]
[image]
[image]
[image]
[nwk_str.txt](https://github.com/project-chip/connectedhomeip/files/7888285/nwk_str.txt)
[security_key.txt](https://github.com/project-chip/connectedhomeip/files/7888286/security_key.txt)
[wificommand.txt](https://github.com/project-chip/connectedhomeip/files/7888287/wificommand.txt)
[wifiversion.txt](https://github.com/project-chip/connectedhomeip/files/7888288/wifiversion.txt)

cecille, 2022-02-01:
Is this on all devices or a specific device?

manjunathambiger23, 2022-02-02:
@cecille we ran this on linux have not tested on other platforms

bzbarsky-apple, 2022-02-04:
Test is trying to read the cluster on endpoint 1, which is not where this cluster lives.

That said, we probably need to merge https://github.com/project-chip/connectedhomeip/pull/14755 for this to really work right.

bzbarsky-apple, 2022-02-05:
And also, ResetCounts might just not work on some platforms.....

kowsisoundhar12, 2022-05-06:
Testscript name ---> [TC-WIFIDIAG-1.3] Command received functionality with server as DUT

ResetCounts Command is giving failure response ,because of this command following attributes response value is not resetting to "0"

BeaconLostCount,
 BeaconRxCount, 
PacketMulticastRxCount,
 PacketMulticastTxCount, 
PacketUnicastRxCount, 
[image]
PacketUnicastTxCount.

kowsisoundhar12, 2022-05-06:
Testscript name ---> [TC-WIFIDIAG-1.3] Command received functionality with server as DUT

ResetCounts Command is giving failure response ,because of this command following attributes response value is not resetting to "0"

BeaconLostCount,
BeaconRxCount,
PacketMulticastRxCount,
PacketMulticastTxCount,
PacketUnicastRxCount,
PacketUnicastTxCount.
[image]
[image]
[WIFIDIAG-Exceution-logs.zip](https://github.com/project-chip/connectedhomeip/files/8641944/WIFIDIAG-Exceution-logs.zip)

bzbarsky-apple, 2022-05-07:
@kowsisoundhar12 getting a nonzero number there is absolutely correct.  See https://github.com/CHIP-Specifications/chip-test-plans/issues/941

kowsisoundhar12, 2022-05-09:
Following attributes are giving failure response 
1.BeaconLostCount
2.PacketMulticastRxCount
3.PacketMulticastTxCount
4.PacketUnicastRxCount
5.PacketUnicastTxCount
6.CurrentMaxRate
7.OverrunCount

Error Screenshot:
[image]

please refer testlogs here
[TC-WIFI-DIAG.zip](https://github.com/project-chip/connectedhomeip/files/8653939/TC-WIFI-DIAG.zip)

bzbarsky-apple, 2022-05-11:
That's running against what server?

kowsisoundhar12, 2022-05-11:
i have executed this  testcase in linux,i have used following commands

1 ) ./scripts/run_in_build_env.sh \
                     "./scripts/build/build_examples.py \
                        --target linux-x64-chip-tool-no-ble-tsan-clang \
                        --target linux-x64-all-clusters-test-group-no-ble-tsan-clang \
                        --target linux-x64-lock-no-ble-tsan-clang \
                        --target linux-x64-tv-app-no-ble-tsan-clang \
                        build \
                        --copy-artifacts-to objdir-clone \
                     "				 
						 
.2) /scripts/run_in_build_env.sh \
                  "./scripts/tests/run_test_suite.py \
                     --chip-tool ./out/linux-x64-chip-tool-no-ble-tsan-clang/chip-tool \
                     run \
                     --iterations 1 \
                     --all-clusters-app ./out/linux-x64-all-clusters-test-group-no-ble-tsan-clang/chip-all-clusters-app \
                     --lock-app ./out/linux-x64-lock-no-ble-tsan-clang/chip-lock-app \
                     --tv-app ./out/linux-x64-tv-app-no-ble-tsan-clang/chip-tv-app \
                  "

bzbarsky-apple, 2022-05-12:
OK, but then how would it pass?  The server is not directly connected to wi-fi, right?

bzbarsky-apple, 2022-06-13:
@manjunath-grl Is there still an issue here?

manjunath-grl, 2022-06-17:
@bzbarsky-apple it is resolved and i tried it in ToT its passing ,can you please close this issue

closed 2022-06-17 by bzbarsky-apple