project-chip/connectedhomeip #13648: Attributes and commands of "Ethernet Network Diagnostics" cluster failing

manjunathambiger23, 2022-01-18:
Summary: 
               Attributes and commands of Ethernet Network Diagnostics cluster  showing failure response when made a run.
Attributes:
PacketRxCount, PacketTxCount, TxErrCount, CollisionCount, OverrunCount, CarrierDetect, TimeSinceReset 
commands:
ResetCounts

Spec screenshot:
[image]

Execution log and screenshot:
[image]
[ethernet_Diag.txt](https://github.com/project-chip/connectedhomeip/files/7889574/ethernet_Diag.txt)

bzbarsky-apple, 2022-01-19:
@yufengwangca

kowsisoundhar12, 2022-05-06:
Testscript name ---> [TC-ETHDIAG-1.2] Command received functionality with server as DUT

ResetCounts Command is giving failure response ,because of this command following attributes response value is not resetting to "0"

1.PacketRxCount,
2.PacketTxCount
3. TxErrCount
4. CollisionCount
5. OverrunCount attribute
[image]
[image]
[Uploading ETHDIAG-2-1-EXecution-logs.zip…]()

naveenkommareddi, 2022-05-17:
this is a DUP of [18321](https://github.com/project-chip/connectedhomeip/issues/18321)

franck-apple, 2022-05-17:
dupe of #18321. Closing

closed 2022-05-17 by franck-apple