project-chip/connectedhomeip #13652: Reference OTA Provider flexible image selection configuration logic (more than 1 version)

hrajendran-samsung, 2022-01-18:
#### Problem
The OTA provider reference application currently provides only one image for OTA-download. This should be augmented with the ability for the OTA-provider to lookup a list of available ota-images and make a selection based on the QueryImage request parameters.
 #### Proposed Solution
Section 11.20.3.3, “Availability of Software Images"
Section 11.20.3.3.2, “Conceptual algorithm for matching OTA Software Images applicable to a query"

The OTA-Provider will implement a solution based on the above sections of the spec.
1. The OTA provider will be provided with a list of software images following the DeviceSoftwareVersionModel Schema. This will be in the form of a simple CSV/JSON file (this can be replaced with a DCL server connection in the future).
2. The OTA provider will process the QueryImage parameters (VendorID, ProductID, and SoftwareVersion)and make a decision which OTA image is best suited from the available list (Section 11.20.3.3.2).
3. The selected OTA file will be sent to the OTA-requestor over the BDX protocol. If no relevant image is found, a "NotAvailable" status will be sent as a response to the QueryImage request.

hrajendran-samsung, 2022-01-18:
Assigning to @hrajendran-samsung

hrajendran-samsung, 2022-01-18:
@selissia @carol-apple If this implementation sounds good, I can go ahead with it.

carol-apple, 2022-03-10:
This has been addressed by https://github.com/project-chip/connectedhomeip/pull/13890

closed 2022-03-10 by carol-apple