project-chip/connectedhomeip #13711: PairingSession::GetSessionType returns incorrect result

mlepage-google, 2022-01-19:
#### Problem
During commissioning, by the time SessionManager::NewPairing (in SessionManager.cpp)
calls PairingSession::GetSessionType to pass it into SecureSessionTable::CreateNewSecureSession,
the value is not set.

You can see this by logging the value and the PairingSession pointer at that location, it's 0 (undefined), not 1 (PASE) or 2 (CASE).

You can add more logging of that value and this pointer in places (e.g. PairingSession::SetSecureSessionType, called by PASESession constructor) to see it was set.

Adding logging of that value and this pointer in PairingSession::Clear reveals it is cleared multiple times between being set and being read, so this is probably why it's not returning the correct result when finally used.

 #### Proposed Solution
Need to determine what the value is used for, whether it matters that it becomes unset, why Clear is called so frequently, etc.

mlepage-google, 2022-01-20:
@emargolis is it as simple as Clear should not reset the session **type**, since that shouldn't change?

emargolis, 2022-01-20:
@mlepage-google I agree with you assessment that Clear() shouldn't reset the session type.

It is fixed in #13772

closed 2022-01-25 by andy31415