project-chip/connectedhomeip #13746: The RendezvousInformation flag generated in manual code and QR code is invalid.

sowmyassp, 2022-01-20:
#### Problem
1) The 'RendezvousInformation' flag generated in the manual code  is  '0'. But according to the spec 0 means device can be discovered via softAP, since SoftAP is not in V1.0 scope, so this value is invalid.

[image]

2) The 'RendezvousInformation' flag generated in the QR code  is  '2' when we launch all-clusters-app using --wifi option which is invalid, as according to the spec the expected value should be  1 while launching the all-clusters-app using --wifi option. 

[image]

Please find the attached DUT log file

[DUT_log.txt](https://github.com/project-chip/connectedhomeip/files/7902961/DUT_log.txt)

dhrishi, 2022-02-01:
@PSONALl PTAL

PSONALl, 2022-02-03:
@sowmyassp As per the specs RendezvousInformation flag is set for modes: 
1  -> SoftAp
2  -> BLE
4  -> on/network

PSONALl, 2022-02-04:
@sowmyassp I think there is quite a confusion here, the values of Rendezvous Information flags are correct in the code as per the specs, and 0 in case of manual pairing represents that the Device does not support any method for rendezvous.
and you can send --capabilities argument while running the application set to the above-mentioned values of Rendezvous flags.  In the second case If the Rendezvous Information flag is set to 2 then the device must be using BLE for discovering

PSONALl, 2022-02-04:
> can send --capabilities argument while running the application set to the above-mentioned values of Rendezvous flags

e.g. for setting SoftAp-mode you need to send the command:
 sudo ./chip-all-clusters-app --capabilities 1

dhrishi, 2022-02-07:
@sowmyassp Closing this issue wrt Sonali's comment. Please feel free to re-open in case of anything.

closed 2022-02-07 by dhrishi

cjandhyala, 2022-02-08:
@PSONALl @dhrishi per @sowmyassp 's comment, here is the excerpt from spec, 0 means device supports Soft AP, which is not a POR in V1.0.

[image]

Also for the 2nd scenario, when an all cluster app on raspi is launched with '--wifi' flag, the DUT does BLE advertisement and commissioner is expected to look for BLE advertisement. Since the QR code generated shows Rendezvous flag to 2, it misleads commissioner to look for the device on the IP network. So when the app is launched with this flag , The Rendezvous flag should be 1.

PSONALl, 2022-02-08:
@cjandhyala  in the table above, first column, **bits** indicate which bit is set and not the value of flag and the description explains the meaning when the bit is 0 and 1

Thus,  
 - if bit **'0'**  is set to 1 (i.e, value of flag = 001 = 1) then the device supports SoftAp 
 - if bit **'0'**  is set to 0 (i.e, value of flag = 000 = 0) means the device does not support any rendezvous mode
 - Similarly in the case of --wifi flag when DUT does BLE advertisement when bit **1** is set i.e, the value of Rendezvous flag is = 010 = 2 
So the value generated for the Rendezvous flag in manual code and QR code is correct