project-chip/connectedhomeip #13750: Reading OperationalCredentials cluster, fabric list attribute may crash the server

erjiaqing, 2022-01-20:
#### Problem

In recent CI for PR https://github.com/project-chip/connectedhomeip/pull/13404, cirque tests failed with

```
[1642664853.500022][65:65] CHIP:IM: Received status response, status is 0
[1642664853.500030][65:65] CHIP:DMG: <RE> OnReportConfirm: NumReports = 0
[1642664853.500037][65:65] CHIP:DMG: IM RH moving to [GeneratingReports]
[1642664853.500113][65:65] CHIP:DMG: <RE:Run> Cluster 37, Attribute 2 is dirty
[1642664853.500124][65:65] CHIP:DMG: Reading attribute: Cluster=0x0000_0037 Endpoint=0 AttributeId=0x0000_0002 (expanded=1)
[1642664853.500498][65:65] CHIP:DL: Found the primary Ethernet interface:eth0
[1642664853.500538][65:65] CHIP:DMG: <RE:Run> Cluster 37, Attribute 3 is dirty
[1642664853.500548][65:65] CHIP:DMG: Reading attribute: Cluster=0x0000_0037 Endpoint=0 AttributeId=0x0000_0003 (expanded=1)
[1642664853.500691][65:65] CHIP:DL: Found the primary Ethernet interface:eth0
[1642664853.500707][65:65] CHIP:DMG: <RE:Run> Cluster 37, Attribute 4 is dirty
[1642664853.500714][65:65] CHIP:DMG: Reading attribute: Cluster=0x0000_0037 Endpoint=0 AttributeId=0x0000_0004 (expanded=1)
[1642664853.500864][65:65] CHIP:DL: Found the primary Ethernet interface:eth0
[1642664853.500883][65:65] CHIP:DMG: <RE:Run> Cluster 37, Attribute 5 is dirty
[1642664853.500891][65:65] CHIP:DMG: Reading attribute: Cluster=0x0000_0037 Endpoint=0 AttributeId=0x0000_0005 (expanded=1)
[1642664853.500974][65:65] CHIP:DL: Found the primary Ethernet interface:eth0
[1642664853.500988][65:65] CHIP:DMG: <RE:Run> Cluster 37, Attribute 6 is dirty
[1642664853.500995][65:65] CHIP:DMG: Reading attribute: Cluster=0x0000_0037 Endpoint=0 AttributeId=0x0000_0006 (expanded=1)
[1642664853.501177][65:65] CHIP:DL: Found the primary Ethernet interface:eth0
[1642664853.501197][65:65] CHIP:DMG: <RE:Run> Cluster 37, Attribute 7 is dirty
[1642664853.501205][65:65] CHIP:DMG: Reading attribute: Cluster=0x0000_0037 Endpoint=0 AttributeId=0x0000_0007 (expanded=1)
[1642664853.501214][65:65] CHIP:DMG: <RE:Run> Cluster 37, Attribute 8 is dirty
[1642664853.501221][65:65] CHIP:DMG: Reading attribute: Cluster=0x0000_0037 Endpoint=0 AttributeId=0x0000_0008 (expanded=1)
[1642664853.501232][65:65] CHIP:DMG: <RE:Run> Cluster 37, Attribute fffc is dirty
[1642664853.501239][65:65] CHIP:DMG: Reading attribute: Cluster=0x0000_0037 Endpoint=0 AttributeId=0x0000_FFFC (expanded=1)
[1642664853.501251][65:65] CHIP:DMG: <RE:Run> Cluster 37, Attribute fffd is dirty
[1642664853.501257][65:65] CHIP:DMG: Reading attribute: Cluster=0x0000_0037 Endpoint=0 AttributeId=0x0000_FFFD (expanded=1)
[1642664853.501267][65:65] CHIP:DMG: <RE:Run> Cluster 3c, Attribute 0 is dirty
[1642664853.501275][65:65] CHIP:DMG: Reading attribute: Cluster=0x0000_003C Endpoint=0 AttributeId=0x0000_0000 (expanded=1)
[1642664853.501283][65:65] CHIP:DMG: <RE:Run> Cluster 3c, Attribute 1 is dirty
[1642664853.501289][65:65] CHIP:DMG: Reading attribute: Cluster=0x0000_003C Endpoint=0 AttributeId=0x0000_0001 (expanded=1)
[1642664853.501296][65:65] CHIP:DMG: <RE:Run> Cluster 3c, Attribute 2 is dirty
[1642664853.501299][65:65] CHIP:DMG: Reading attribute: Cluster=0x0000_003C Endpoint=0 AttributeId=0x0000_0002 (expanded=1)
[1642664853.501308][65:65] CHIP:DMG: <RE:Run> Cluster 3c, Attribute fffd is dirty
[1642664853.501315][65:65] CHIP:DMG: Reading attribute: Cluster=0x0000_003C Endpoint=0 AttributeId=0x0000_FFFD (expanded=1)
[1642664853.501322][65:65] CHIP:DMG: <RE:Run> Cluster 3e, Attribute 1 is dirty
[1642664853.501328][65:65] CHIP:DMG: Reading attribute: Cluster=0x0000_003E Endpoint=0 AttributeId=0x0000_0001 (expanded=1)
[1642664853.501345][65:65] CHIP:DMG: Error retrieving data from clusterId: 0x0000_003E, err = ../../src/lib/core/CHIPTLVReader.cpp:746: CHIP Error 0x00000025: Unknown implicit TLV tag
[1642664853.501355][65:65] CHIP:EM: Sending Standalone Ack for MessageCounter:1723582 on exchange 29429r
[1642664853.501374][65:65] CHIP:IN: Prepared secure message 0x7fffffffe220 to 0x000000000001B669 (1)  of type 0x10 and protocolId (0, 0) on exchange 29429r with MessageCounter:4807234.
[1642664853.501387][65:65] CHIP:IN: Sending encrypted msg 0x7fffffffe220 with MessageCounter:4807234 to 0x000000000001B669 (1) at monotonic time: 1897485 msec
[1642664853.501426][65:65] CHIP:EM: Flushed pending ack for MessageCounter:1723582 on exchange 29429r
```

After checking backtrace, this issue is unlikely to be caused by the PR.

yufengwangca, 2022-02-02:
This issue has been assigned, need further investigation, remove v1_triage_split_10 label

erjiaqing, 2022-02-15:
Update: re-enabled the test on forked repo and CI is green, will continue monitor this issue to see if it can be reproduced.

PR #15236 (2022-02-17): Re-enable wildcard read test
#### Problem
Per recent runs, we can nolonger reproduce #13750 

Closes #13750 

#### Change overview
Re-enable the test.

#### Testing
Update the test to see if the server crashes when doing wildcard read.
closed 2022-02-17 by andy31415