project-chip/connectedhomeip #13758: [TC-PS-2.1] Attributes Missing From Power Source Cluster

kowsisoundhar12, 2022-01-20:
Summary:
The following  attributes seem to be missing in the Power Source Cluster
	1.WiredAssessedInputVoltage
	2.WiredAssessedInputFrequency
	3.WiredCurrentType
	4.WiredAssessedCurrent
	5.WiredNominalVoltage
	6.WiredMaximumCurrent
	7.WiredPresent
	8.ActiveWiredFaults
	9.BatReplacementNeeded
	10.BatReplaceability
	11.BatPresent
	12.BatReplacementDescription
	13.BatCommonDesignation
	14.BatANSIDesignation
	15.BatIECDesignation
	16.BatApprovedChemistry
	17.BatCapacity
	18.BatQuantity
	19.BatTimeToFullCharge
	20.BatFunctionalWhileCharging
	21.BatChargingCurrent
	22.ActiveBatChargeFaults
Spec-screenshot:
[image]
[image]

Testscripts Logs:
[image]

bzbarsky-apple, 2022-05-05:
chip-tool should support these now.  @kowsisoundhar12 can you confirm?

kowsisoundhar12, 2022-05-05:
@bzbarsky-apple still observing above attribute missing error in power Source Cluster
[image]

bzbarsky-apple, 2022-05-05:
Ah, yaml is still using controller-clusters.  Thanks, I'll get a PR up.

closed 2022-05-07 by bzbarsky-apple