project-chip/connectedhomeip #13787: Android ZAP generator produces bad code when command has nullable structure as an argument

Morozov-5F, 2022-01-20:
#### Problem
Door Lock cluster has command Clear Credential which has Nullable structure as an argument. Generator for `CHIPClusters-JNI.cpp` does not behave well in that case and produces clearly wrong code which uses undeclared parameters:

```c++
JNI_METHOD(void, DoorLockCluster, clearCredential)
(JNIEnv * env, jobject self, jlong clusterPtr, jobject callback, jobject credentialType, jobject credentialIndex, jobject timedInvokeTimeoutMs)
{
    chip::DeviceLayer::StackLock lock;
    CHIP_ERROR err = CHIP_NO_ERROR;
    DoorLockCluster * cppCluster;

    chip::app::Clusters::DoorLock::Commands::ClearCredential::Type request;

    request.credential = credential == nullptr
        ? chip::app::DataModel::Nullable<chip::app::Clusters::DoorLock::Structs::DlCredential::Type>()
        : chip::app::DataModel::Nullable<chip::app::Clusters::DoorLock::Structs::DlCredential::Type>(credentialValue);

    // And so on . . .
```

It seems to me that the generator creates a new function with parameters that correspond to the structure field, however the code expects to have a pointer to structure as an argument.

 #### Proposed Solution
I'm not sure how to fix this. Maybe for such cases we don't need to expand structure fields into the individual parameters but leave the structure as a whole?

austinh0, 2022-01-20:
#13609 reworks the codegen for structs, is it fixed in that PR?

Morozov-5F, 2022-01-26:
Yep, #13609 fixes that.

closed 2022-01-26 by Morozov-5F