project-chip/connectedhomeip #13823: Basic Location returns default "$"

yinyihu-silabs, 2022-01-21:
When reading basic cluster location attribute, it returns "$", invalid country code nor default "XX".
`chip-tool basic read location 2954 0`
```CHIP:EM: Removed CHIP MessageCounter:15025051 from RetransTable on exchange 43789i
CHIP:DMG: ReportDataMessage =
CHIP:DMG: {
CHIP:DMG: 	AttributeReportIBs =
CHIP:DMG: 	[
CHIP:DMG: 		AttributeReportIB =
CHIP:DMG: 		{
CHIP:DMG: 			AttributeDataIB =
CHIP:DMG: 			{
CHIP:DMG: 				DataVersion = 0x0,
CHIP:DMG: 				AttributePathIB =
CHIP:DMG: 				{
CHIP:DMG: 					Endpoint = 0x0,
CHIP:DMG: 					Cluster = 0x28,
CHIP:DMG: 					Attribute = 0x0000_0006,
CHIP:DMG: 				}
CHIP:DMG: 					
CHIP:DMG: 					Data = "$", 
CHIP:DMG: 			},
CHIP:DMG: 			
CHIP:DMG: 		},
CHIP:DMG: 		
CHIP:DMG: 	],
CHIP:DMG: 	
CHIP:DMG: 	SuppressResponse = true, 
CHIP:DMG: }
CHIP:TOO: Basic.Location response: $
CHIP:EM: Sending Standalone Ack for MessageCounter:14594985 on exchange 43789i
CHIP:IN: Prepared secure message 0xffffa77fc9e8 to 0x0000000000000B8A (1)  of type 0x10 and protocolId (0, 0) on exchange 43789i with MessageCounter:15025052.
CHIP:IN: Sending encrypted msg 0xffffa77fc9e8 with MessageCounter:15025052 to 0x0000000000000B8A (1) at monotonic time: 504705951 msec
CHIP:EM: Flushed pending ack for MessageCounter:14594985 on exchange 43789i

lzgrablic02, 2022-01-31:
@yinyihu-silabs what build/device is the server?

yinyihu-silabs, 2022-01-31:
I'm running lighting/lock/window app on efr32.  All apps returns the same $ value as default.
After write, read would return expected country code.

lzgrablic02, 2022-02-02:
I guess this might be a platform-specific issue, then. Unassigning myself.

bzbarsky-apple, 2022-05-05:
@yinyihu-silabs Is this still an issue on current tip?  I would expect this to work unless the configuration manager is returning a bogus value...

closed 2022-05-09 by yinyihu-silabs