project-chip/connectedhomeip #13824: SoftwareDiagnostics reset-watermarks doesn't reset

yinyihu-silabs, 2022-01-21:
#### Problem
 CurrentHeapHighWatermark remains the same before and after `chip-tool softwarediagnostics reset-watermarks` command.

jepenven-silabs, 2022-07-22:
Fixed : https://github.com/project-chip/connectedhomeip/pull/19104

closed 2022-07-22 by jepenven-silabs

yinyihu-silabs, 2022-09-06:
Issue is still there.
```
ubuntu@ubuntu:~$ ./chip-tool softwarediagnostics  reset-watermarks 1 0
[1662490793.763453][4651:4651] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_kvs
[1662490793.769030][4651:4651] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_factory.ini
[1662490793.769319][4651:4651] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_config.ini
[1662490793.769460][4651:4651] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_counters.ini
[1662490793.769839][4651:4651] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-LJhdg3)
[1662490793.770511][4651:4651] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1662490793.770584][4651:4651] CHIP:DL: NVS set: chip-counters/reboot-count = 22 (0x16)
[1662490793.771684][4651:4651] CHIP:DL: Got Ethernet interface: eth0
[1662490793.772388][4651:4651] CHIP:DL: Found the primary Ethernet interface:eth0
[1662490793.773176][4651:4651] CHIP:DL: Got WiFi interface: wlan0
[1662490793.779322][4651:4651] CHIP:DL: Found the primary WiFi interface:wlan0
[1662490793.779406][4651:4651] CHIP:IN: UDP::Init bind&listen port=0
[1662490793.779523][4651:4651] CHIP:IN: UDP::Init bound to port=46569
[1662490793.779546][4651:4651] CHIP:IN: BLEBase::Init - setting/overriding transport
[1662490793.779566][4651:4651] CHIP:IN: TransportMgr initialized
[1662490793.779600][4651:4651] CHIP:FP: Initializing FabricTable from persistent storage
[1662490793.779753][4651:4651] CHIP:TS: Last Known Good Time: 2022-09-01T18:26:37
[1662490793.781231][4651:4651] CHIP:FP: Fabric index 0x1 was retrieved from storage. Compressed FabricId 0x92AA20E93FC9E286, FabricId 0x0000000000000001, NodeId 0x000000000001B669, VendorId 0xFFF1
[1662490793.783507][4651:4651] CHIP:ZCL: Using ZAP configuration...
[1662490793.788279][4651:4651] CHIP:DL: Avahi client registered
[1662490793.789942][4651:4651] CHIP:CTL: System State Initialized...
[1662490793.790060][4651:4651] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1662490793.790116][4651:4651] CHIP:CTL: Setting attestation nonce to random value
[1662490793.790165][4651:4651] CHIP:CTL: Setting CSR nonce to random value
[1662490793.790566][4651:4656] CHIP:DL: CHIP task running
[1662490793.790866][4651:4656] CHIP:DL: HandlePlatformSpecificBLEEvent 32784
[1662490793.790982][4651:4656] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1662490793.791035][4651:4656] CHIP:CTL: Setting attestation nonce to random value
[1662490793.791274][4651:4656] CHIP:CTL: Setting CSR nonce to random value
[1662490793.792529][4651:4656] CHIP:CTL: Generating NOC
[1662490793.793865][4651:4656] CHIP:FP: Validating NOC chain
[1662490793.796495][4651:4656] CHIP:FP: NOC chain validation successful
[1662490793.796755][4651:4656] CHIP:FP: Updated fabric at index: 0x1, Node ID: 0x000000000001B669
[1662490793.796802][4651:4656] CHIP:TS: Last Known Good Time: 2022-09-01T18:26:37
[1662490793.796836][4651:4656] CHIP:TS: New proposed Last Known Good Time: 2021-01-01T00:00:00
[1662490793.796865][4651:4656] CHIP:TS: Retaining current Last Known Good Time
[1662490793.846407][4651:4656] CHIP:FP: Metadata for Fabric 0x1 persisted to storage.
[1662490793.850220][4651:4656] CHIP:TS: Committing Last Known Good Time to storage: 2022-09-01T18:26:37
[1662490793.853867][4651:4656] CHIP:CTL: Joined the fabric at index 1. Compressed fabric ID is: 0x0000000000000000
[1662490793.863896][4651:4656] CHIP:TOO: Sending command to node 0x1
[1662490793.863948][4651:4656] CHIP:CSM: FindOrEstablishSession: PeerId = [1:0000000000000001]
[1662490793.863971][4651:4656] CHIP:CSM: FindOrEstablishSession: No existing OperationalSessionSetup instance found
[1662490793.864006][4651:4656] CHIP:CTL: OperationalSessionSetup[1:0000000000000001]: State change 1 --> 2
[1662490793.864035][4651:4656] CHIP:DIS: Resolving 92AA20E93FC9E286:0000000000000001 ...
[1662490793.972067][4651:4656] CHIP:DL: Avahi resolve found
[1662490793.972223][4651:4656] CHIP:DIS: Node ID resolved for 92AA20E93FC9E286:0000000000000001
[1662490793.972288][4651:4656] CHIP:DIS:        Hostname: EA1FEB7E88E978B2
[1662490793.972349][4651:4656] CHIP:DIS:        IP Address #1: fd11:35::a106:40e7:f142:8d6
[1662490793.972410][4651:4656] CHIP:DIS:        Port: 5540
[1662490793.972532][4651:4656] CHIP:DIS:        Mrp Interval idle: 5000 ms
[1662490793.972644][4651:4656] CHIP:DIS:        Mrp Interval active: 2000 ms
[1662490793.972805][4651:4656] CHIP:DIS:        TCP Supported: 0
[1662490793.974728][4651:4656] CHIP:DIS: Lookup clearing interface for non LL address
[1662490793.974814][4651:4656] CHIP:DIS: UDP:[fd11:35::a106:40e7:f142:8d6%veth9a68622]:5540: new best score: 6
[1662490793.974872][4651:4656] CHIP:DIS: Checking node lookup status after 111 ms
[1662490793.974925][4651:4656] CHIP:DIS: Keeping DNSSD lookup active
[1662490794.064159][4651:4656] CHIP:DIS: Checking node lookup status after 200 ms
[1662490794.064228][4651:4656] CHIP:DIS: OperationalSessionSetup[1:0000000000000001]: Updating device address to UDP:[fd11:35::a106:40e7:f142:8d6]:5540 while in state 2
[1662490794.064251][4651:4656] CHIP:CTL: OperationalSessionSetup[1:0000000000000001]: State change 2 --> 3
[1662490794.064381][4651:4656] CHIP:IN: SecureSession[0xffff7c00a0a0]: Allocated Type:2 LSID:10351
[1662490794.064426][4651:4656] CHIP:SC: Initiating session on local FabricIndex 1 from 0x000000000001B669 -> 0x0000000000000001
[1662490794.065277][4651:4656] CHIP:SC: Including MRP parameters
[... cut ...]
[1662490808.337952][4664:4664] CHIP:BLE: BleConnectionDelegate::CancelConnection is not implemented.
[1662490808.338707][4664:4664] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-lOme0c)
[1662490808.339630][4664:4664] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1662490808.339719][4664:4664] CHIP:DL: NVS set: chip-counters/total-operational-hours = 0 (0x0)
[1662490808.339745][4664:4664] CHIP:DL: Inet Layer shutdown
[1662490808.339764][4664:4664] CHIP:DL: BLE shutdown
[1662490808.339781][4664:4664] CHIP:DL: System Layer shutdown
ubuntu@ubuntu:~$
```