project-chip/connectedhomeip #13825: Self commissioning must bootstrap IPK; commissioners should distribute to commissionees by default

msandstedt, 2022-01-21:
#### Problem

Self-commissioned nodes (commissioners and controllers) have no mechanism to bootstrap their IPK(s) and no mechanism to distribute these to commissionees if no IPK is provided by singing infrastructure.

 #### Proposed Solution

#13503 introduces an optional interface for the operational credentials delegate to specify an IPK along with certificates.

If provided by the credentials delegate, self-commissioning nodes (commissioners and controllers) should store this during self-commissioning and use it subsequently.  If not provided, these nodes must generate one randomly.

If desired, the signer-distribution pathway could enclose all of past / current / future IPK epoch keys, but the minimum pathway is distribution of the current key.

Subsequently, when commissioning other nodes, commissioners should distribute their current IPK by enclosing it in the AddNOC command if no IPK was provided by signing infrastructure.

cecille, 2022-03-10:
@msandstedt - there is a merged PR in the linked issue. Is there work remaining on this issue?

msandstedt, 2022-03-29:
@cecille , 

Yes, https://github.com/project-chip/connectedhomeip/pull/16737 delivers most of the IPK functionality.  But this is still needed.

cecille, 2022-03-29:
Adding a couple of folks here that are looking at IPK stuff

msandstedt, 2022-03-29:
To add some context, post-https://github.com/project-chip/connectedhomeip/pull/16737, it is my view that the sdk will be in a place where it now allows ecosystems to implement end-to-end use of IPKs.  Certainly, we will no longer be blocked once this is merged.

But there will be no in-tree example that uses and distributes non-default IPKs.

Maybe we are OK with that.  I don't see that as a critical thing for feature cut.

cecille, 2022-04-01:
I would consider this feature complete, but we do need to verify non-default IPKs as part of the testing. The PR mentions a followup allowing this to be settable through chip-tool, which would be a good example. However, I think we can very reasonably say that work can be completed post-4/4, during the weeks where we are building out the testing. Does that work for you?

msandstedt, 2022-04-01:
Yes, completely agree.

msandstedt, 2022-04-13:
Per discussion, this is resolved with our current code in master.

closed 2022-04-13 by msandstedt