project-chip/connectedhomeip #13827: Need a reset command to avoid launching the DUT each time while pairing with TH

sowmyassp, 2022-01-21:
#### Problem

Please provide the reset command which can avoid launching the DUT each time while pairing with TH.

Paring command:
On DUT  : sudo ./chip-all-clusters-app --wifi 
On TH  : ./chip-tool pairing ble-wifi 1 zigbee matter123 20202021 3840

woody-apple, 2022-01-21:
Proposed workaround for v1, is to delete the chip storage file. @bzbarsky-apple Can you update this issue with the location, and we can resolve this.

bzbarsky-apple, 2022-01-25:
@sowmyassp You could remove the already-commissioned fabric, no?  That should put the device back into the "no fabrics are commissioned" state...

Now whether we handle that state transition right, I don't know.  But I'd start with testing that, since that's the spec-defined way to do what you want.

(Unless you just want to commission it into a second fabric, in which case you open a commissioning window.)

woody-apple, 2022-01-26:
Looks like this is possible, closing given @bzbarsky-apple's comments, please re-open if needed

closed 2022-01-26 by woody-apple