project-chip/connectedhomeip #13842: [TC-MF-1.17] Aborted message during commissioning process

praveenCY, 2022-01-22:
#### Problem
Below error log during opening commissioning window second time (TC-MF-1.17)

[1642730306.013271][3703:3703] CHIP:SPT: VerifyOrDie failure at ../../src/messaging/ExchangeMgr.cpp:102: false
Aborted (core dumped) 

#### Steps:

- TH_CR1 start commission with DUT
- TH_CR1 open a commissioning window and TH_CR2 commission with DUT_CE
- TH_CR1 open a commissioning window 

==> It fails here 

Logs: [LOCK-TC-MF-1.17.txt](https://github.com/project-chip/connectedhomeip/files/7917342/LOCK-TC-MF-1.17.txt)

holbrookt, 2022-01-31:
@praveenCY is this happening on a specific platform? Please add a label if so

praveenCY, 2022-02-01:
> @praveenCY is this happening on a specific platform? Please add a label if so

this is seen on Infineon P6 platform

holbrookt, 2022-02-01:
Thanks @praveenCY. There doesn't seem to be an infineon or p6 label

andy31415, 2022-02-07:
I believe that we do not handle the error-case well in chip-tool.

The logs show a lot of `[1642657572.512396][4785:4790] CHIP:CTL: Device connection failed. Error ../../src/app/OperationalDeviceProxy.cpp:156: CHIP Error 0x0000000B: No memory` which is likely the cause. The core dump is just a final symptom

bzbarsky-apple, 2022-06-13:
@praveenCY Is this still reproducible?

closed 2022-12-21 by stale[bot]