project-chip/connectedhomeip #13843: Device will crash when receiving and reply the mDNS packets during pairing with chip-tool or android app

jamesluo11, 2022-01-22:
#### Problem
Device will crash when receiving and reply the mDNS packets during pairing with chip-tool or android app
i guess the issue results from the lwip pbuf management.

My platform is ARM + FreeRTOS + Lwip.  and i choose the minimal mDNS policy,
After getting the ip, device starts the DNS server and start to listen 5353 port,  during receiving and sending reply mDNS packets, the device will crash with data abort. i see the lr register point to mem_malloc. 

Firstly in general idea , in the upstream, device Mac layer allocate the pbuf and Matter will free it by PacketBufferHandle after handling it within lambda event. in the downstream, Matter will allocate the pbuf by PacketBufferHandle  and device Mac layer will free it after sending it out.

After the analysis, i found the memory heap in lwip is messed(i use the default policy memory management in lwip), when i enable the sanity check in lwip, i can find "mem_free: illegal memory: non-linked: double free".
And i have checked the synchronization mechanism in lwip, i found it works as expected, it protects the mem_malloc and mem_free with mutex.

BTW, when i comment out the Lambda event handler,  the device will NOT crash even the commission is incomplete.

 #### Proposed Solution
Still can't find root cause. i assume the lwip can handle the concurrent memory access.

jamesluo11, 2022-01-22:
@bzbarsky-apple @tcarmelveilleux  i found the root cause just now.  In device Mac layer, in upstream we should  call pbuf_alloc(PBUF_RAW, len, **PBUF_POOL**)  rather than buf_alloc(PBUF_RAW, len, **PBUF_RAM**);  What enlightens me is i notice that PacketBufferHandle allocate pbuf with PBUF_POOL.
It's a little tricky.

kpschoedel, 2022-01-24:
I'll make the `pbuf_type` a build configuration option, defaulting to `PBUF_RAM`.

jamesluo11, 2022-01-25:
@kpschoedel I think PBUF_POOL is the default value, **keep the current design**. And leave some comment there, hope it can catch the attention.

tcarmelveilleux, 2022-01-27:
> @bzbarsky-apple @tcarmelveilleux i found the root cause just now. In device Mac layer, in upstream we should call pbuf_alloc(PBUF_RAW, len, **PBUF_POOL**) rather than buf_alloc(PBUF_RAW, len, [...]

I disagree that this is a root cause. The location of where we allocated SHOULD NOT impact this bug. A double-free is a double-free. It's possible the local change of PBUF_POOL to PBUF_RAM slightly affected the code paths or timing, but it is unlikely the root cause

jamesluo11, 2022-03-14:
> 

Yeah, these days i want to make the root cause clear,  and i change the pbuf type to PBUF_RAM in the phy layer when allocating ram to convey packets to lwip layer,  the crash still happen always.(Before i just use PBUF_POOL for walk-around)
And i think it should be free for developers to use PBUF_RAM or PBUF_POOL to convey packets to upper layer. i still can't find the root cause.

Do you have any idea?

bzbarsky-apple, 2022-03-14:
> And i think it should be free for developers to use PBUF_RAM or PBUF_POOL to convey packets to upper layer.

Sure, but given that the upper layer is then supposed to free the packet and the packet does not itself know how it was allocated, that means the upper layer needs to be configured in a way that matches the lower layer in terms of what sort of allocation strategy is used, right?

jamesluo11, 2022-03-15:
> Sure, but given that the upper layer is then supposed to free the packet and the packet does not itself know how it was allocated, that means the upper layer needs to be configured in a way that [...]

I think Matter SDK just use pbuf_free(), which internally can do the right thing according to the some information in the pbuf.

jamesluo11, 2022-03-15:
> packet does not itself know how it was allocated

@tcarmelveilleux  Sorry , can you explain a little more? I assume pbuf internal has enough information to record how it organises

bzbarsky-apple, 2022-03-16:
Ah, we just call `pbuf_free` and internally it does in fact track the type.  So yes, what @tcarmelveilleux said about this just being a double-free seems correct then, since "mixing" pbuf types should work fine.

jamesluo11, 2022-03-16:
> Ah, we just call `pbuf_free` and internally it does in fact track the type. So yes, what @tcarmelveilleux said about this just being a double-free seems correct then, since "mixing" pbuf types should work fine.

@bzbarsky-apple Thanks, i agree that double-free cause this issue. But i have strictly implemented the synchronous mechnism(sys_arch.c based on FreeRTOS), i assume the lwip can guarantee the race of resource no matter we call pbuf_calloc and pbuf_free in whatever thread context. So i can't understand what caused the double-free. And the issue only happen when i use PBUF_RAM in mac layer. In other project, i always use PBUF_RAM , there is no problem.
(I observe that once Matter SDK broadcast the mDNS packets, the issue occur immediately. )

closed 2022-12-21 by stale[bot]