project-chip/connectedhomeip #13864: chip-tool unable to establish the CASE session with Silicon Labs EFR32 DUT

lds-jinfeng, 2022-01-24:
#### Problem
I am using Silicon Labs EFR32MG12P as DUT and chip-tool as TH, my DUT is able to be configured to Nordic OTBR's network, but I am unable to control my DUT. For details of this issue, please see attached logs.

[TH-pairing-control-ping-0124.log](https://github.com/project-chip/connectedhomeip/files/7925691/TH-pairing-control-ping-0124.log)
[Nordic-RCP-ot-br-posix-git-version-info.txt](https://github.com/project-chip/connectedhomeip/files/7925709/Nordic-RCP-ot-br-posix-git-version-info.txt)
[DUT-pairing-ping-0124.log](https://github.com/project-chip/connectedhomeip/files/7925693/DUT-pairing-ping-0124.log)
[DUT-git-version-info.txt](https://github.com/project-chip/connectedhomeip/files/7925695/DUT-git-version-info.txt)

DUT IP addr got from serial connection.
> ipaddr
fd48:c45a:4b85:9b7:0:ff:fe00:2c00
fd4a:66f1:5e91:3010:9b8b:ded5:9cd7:2ffa
fd48:c45a:4b85:9b7:9b0e:b4f2:2eb8:be6f
fe80:0:0:0:e4f4:4f1a:e68d:5649
Done

 #### Proposed Solution
None.

lds-jinfeng, 2022-01-25:
Hi All,
Sorry for my previous errors, I used a wrong node id for command ./chip-tool onoff toggle <node-id> <endpoint id>, today I retry with correct node id, TH is able to control my DUT, but after a few minutes, TH become unable to control my DUT, ping is OK. See below logs for details.

[TH-pairing-control-ping-0125.txt](https://github.com/project-chip/connectedhomeip/files/7930408/TH-pairing-control-ping-0125.txt)
[DUT-pairing-control-ping-0125.txt](https://github.com/project-chip/connectedhomeip/files/7930409/DUT-pairing-control-ping-0125.txt)

DUT IP addr info (DUT factory rest and pairing again, so its IP addr changed)
ipaddr
fd48:c45a:4b85:9b7:0:ff:fe00:b000
fd4a:66f1:5e91:3010:4e24:d8ab:e5e1:6371
fd48:c45a:4b85:9b7:1f2a:1ac0:b81c:460c
fe80:0:0:0:985b:2f6:d7e8:4192
Done

jmartinez-silabs, 2022-01-25:
It is hard to say what happened.

```
<info  > [IN] Sending encrypted msg 0x2001c1bc with MessageCounter:13793544 to 0x000000000001B669 (1) at monotonic time: ld msec
<detail> [DL] Thread packet RCVD: UDP, len 66
<detail> [DL]     src  F<detail> [DL] Thread packet SENT: UDP, len 112
<detail> [DL]     src  FD4A:66F1:5E91:3010:4E24:D8AB:E5E1:6371, port 5540
<detail> [DL]     dest FD4A:66F1:5E91:3010:F69C:79AA:7646:4C6B, port 5542
<detail> [DMG] ICR moving to [CommandSen]
<detail> [DMG] ICR moving to [AwaitingDe]
<info  > [EM] Received message of type 0xhx with protocolId (16, 0) and MessageCounter:0 on exchange 36201?
<detail> [EM] Found matching exchange: 43938r, Delegate: 0
<detail> [EM] Rxd Ack; Removing MessageCounter:2379158691 f<efr32 > Light ON
<detail> [DL] Thread packet RCVD: UDP, len 82
<detail> [DL]     src  FD4A:66F1:5E91:3010:F69C:79AA:7646:4C6B, port 5542
<detail> [DL]     dest FD4A:66F1:5E91:3010:4E24:D8AB:E5E1:6371, port 5540
<info  > [EM] Received message of type 0xhx with protocolId (16, 0) and MessageCounter:0 on exchange 10803?
<detail> [EM] Found matching exchange: 43939r, Delegate: 0
<detail> [EM] Rxd Ack; Removing MessageCounter:13793544 from Retrans Table on exchange 43939r
<detail> [EM] Removed CHIP MessageCounter:13793544 from RetransTable on exchange 43939r
<detail> [EM] Retransmitting MessageCounter:4368767 on exchange 1032r Send Cnt 0
```

From this device trace. The DUT replied and the CHIP tool seems to miss it. It resends the same message. From there the com seems broken.

I will try to reproduce it. So far was able to send commands to my DUT even a few hours after commissioning it. I do use an EFR32 RCP, I don't know the Nordic RCP OTBR.

turon, 2022-02-01:
These basic flows should be rock solid for v1.
Removing tag `v1_triage_split_4`.

bzbarsky-apple, 2022-05-19:
@lds-jinfeng Is this still reproducible with the TE9 or tip SHA?

bzbarsky-apple, 2022-05-20:
@lds-jinfeng So this can be closed?

lds-jinfeng, 2022-05-23:
@bzbarsky-apple Yes, this case can be closed, since we did not see this issue on TE8 and TE9.

closed 2022-05-23 by bzbarsky-apple