project-chip/connectedhomeip #13879: While testing Thermostat Cluster 3.2.2. [TC-TSTAT-2.2] Setpoint Test Cases with server as DUT test case, getting default error response while doing setpoint-raise-lower

Garik098, 2022-01-24:
While testing  [TC-TSTAT-2.2] with chip tool, for setpoint raise lower command, seeing the default error response of 0xc3. Expected response is not an error response.

./out/debug/chip-tool thermostat setpoint-raise-lower 0 30 1 1

"[1642752351345] [4442:201443] CHIP: [DL] _Init
[1642752351346] [4442:201443] CHIP: [IN] UDP::Init bind&listen port=5542
[1642752351347] [4442:201443] CHIP: [IN] UDP::Init bound to port=5542
[1642752351347] [4442:201443] CHIP: [IN] UDP::Init bind&listen port=5542
[1642752351347] [4442:201443] CHIP: [IN] UDP::Init bound to port=5542
[1642752351347] [4442:201443] CHIP: [IN] TransportMgr initialized
[1642752351347] [4442:201443] CHIP: [DIS] Init fabric pairing table with server storage
[1642752351347] [4442:201443] CHIP: [ZCL] Using ZAP configuration...
[1642752351347] [4442:201443] CHIP: [CTL] System State Initialized...
[1642752351349] [4442:201443] CHIP: [CTL] Generating NOC
[1642752351349] [4442:201443] CHIP: [CTL] Generating ICAC
[1642752351349] [4442:201443] CHIP: [DIS] Verifying the received credentials
[1642752351350] [4442:201443] CHIP: [IN] Generating compressed fabric ID using uncompressed fabric ID 0x0000000000000001 and root pubkey
[1642752351350] [4442:201443] CHIP: [IN] 0x04, 0x37, 0x97, 0x88, 0x1c, 0xe4, 0x6f, 0xf6, 
[1642752351350] [4442:201443] CHIP: [IN] 0x42, 0x3a, 0x48, 0x8b, 0xd4, 0x4b, 0x5c, 0x82, 
[1642752351350] [4442:201443] CHIP: [IN] 0x4f, 0xb9, 0x07, 0x44, 0x9d, 0x8e, 0x7c, 0xc5, 
[1642752351350] [4442:201443] CHIP: [IN] 0x4e, 0xc4, 0x4e, 0x5d, 0x3c, 0x1e, 0x08, 0x5a, 
[1642752351350] [4442:201443] CHIP: [IN] 0xc4, 0xf2, 0x68, 0x24, 0x06, 0x05, 0x78, 0x9d, 
[1642752351350] [4442:201443] CHIP: [IN] 0x08, 0x55, 0x3e, 0x88, 0x87, 0x36, 0x96, 0xba, 
[1642752351350] [4442:201443] CHIP: [IN] 0x0c, 0x36, 0xa5, 0x9b, 0xc2, 0x25, 0xba, 0x11, 
[1642752351350] [4442:201443] CHIP: [IN] 0x4a, 0x7d, 0xc3, 0x35, 0x74, 0x8f, 0x9a, 0x02, 
[1642752351350] [4442:201443] CHIP: [IN] 0xca, 
[1642752351350] [4442:201443] CHIP: [IN] Generated compressed fabric ID
[1642752351350] [4442:201443] CHIP: [IN] 0x18, 0x40, 0x62, 0x9b, 0x9d, 0x0c, 0x3d, 0xd2, 
[1642752351350] [4442:201443] CHIP: [DIS] Added new fabric at index: 1, Initialized: 1
[1642752351350] [4442:201443] CHIP: [DIS] Assigned compressed fabric ID: 0x1840629B9D0C3DD2, node ID: 0x000000000001B669
[1642752351350] [4442:201443] CHIP: [CTL] Joined the fabric at index 1. Compressed fabric ID is: 0x1840629B9D0C3DD2
[1642752351350] [4442:201443] CHIP: [CTL] Generating NOC
[1642752351350] [4442:201443] CHIP: [CTL] Generating ICAC
[1642752351350] [4442:201443] CHIP: [DIS] Verifying the received credentials
[1642752351351] [4442:201443] CHIP: [IN] Generating compressed fabric ID using uncompressed fabric ID 0x0000000000000002 and root pubkey
[1642752351351] [4442:201443] CHIP: [IN] 0x04, 0x37, 0x97, 0x88, 0x1c, 0xe4, 0x6f, 0xf6, 
[1642752351351] [4442:201443] CHIP: [IN] 0x42, 0x3a, 0x48, 0x8b, 0xd4, 0x4b, 0x5c, 0x82, 
[1642752351351] [4442:201443] CHIP: [IN] 0x4f, 0xb9, 0x07, 0x44, 0x9d, 0x8e, 0x7c, 0xc5, 
[1642752351351] [4442:201443] CHIP: [IN] 0x4e, 0xc4, 0x4e, 0x5d, 0x3c, 0x1e, 0x08, 0x5a, 
[1642752351351] [4442:201443] CHIP: [IN] 0xc4, 0xf2, 0x68, 0x24, 0x06, 0x05, 0x78, 0x9d, 
[1642752351351] [4442:201443] CHIP: [IN] 0x08, 0x55, 0x3e, 0x88, 0x87, 0x36, 0x96, 0xba, 
[1642752351351] [4442:201443] CHIP: [IN] 0x0c, 0x36, 0xa5, 0x9b, 0xc2, 0x25, 0xba, 0x11, 
[1642752351351] [4442:201443] CHIP: [IN] 0x4a, 0x7d, 0xc3, 0x35, 0x74, 0x8f, 0x9a, 0x02, 
[1642752351351] [4442:201443] CHIP: [IN] 0xca, 
[1642752351351] [4442:201443] CHIP: [IN] Generated compressed fabric ID
[1642752351351] [4442:201443] CHIP: [IN] 0xe3, 0xc9, 0xd1, 0x55, 0xfa, 0x55, 0xfe, 0xd7, 
[1642752351351] [4442:201443] CHIP: [DIS] Added new fabric at index: 2, Initialized: 1
[1642752351351] [4442:201443] CHIP: [DIS] Assigned compressed fabric ID: 0xE3C9D155FA55FED7, node ID: 0x000000000001B669
[1642752351351] [4442:201443] CHIP: [CTL] Joined the fabric at index 2. Compressed fabric ID is: 0xE3C9D155FA55FED7
[1642752351352] [4442:201443] CHIP: [CTL] Generating NOC
[1642752351352] [4442:201443] CHIP: [CTL] Generating ICAC
[1642752351352] [4442:201443] CHIP: [DIS] Verifying the received credentials
[1642752351352] [4442:201443] CHIP: [IN] Generating compressed fabric ID using uncompressed fabric ID 0x0000000000000003 and root pubkey
[1642752351352] [4442:201443] CHIP: [IN] 0x04, 0x37, 0x97, 0x88, 0x1c, 0xe4, 0x6f, 0xf6, 
[1642752351352] [4442:201443] CHIP: [IN] 0x42, 0x3a, 0x48, 0x8b, 0xd4, 0x4b, 0x5c, 0x82, 
[1642752351352] [4442:201443] CHIP: [IN] 0x4f, 0xb9, 0x07, 0x44, 0x9d, 0x8e, 0x7c, 0xc5, 
[1642752351352] [4442:201443] CHIP: [IN] 0x4e, 0xc4, 0x4e, 0x5d, 0x3c, 0x1e, 0x08, 0x5a, 
[1642752351352] [4442:201443] CHIP: [IN] 0xc4, 0xf2, 0x68, 0x24, 0x06, 0x05, 0x78, 0x9d, 
[1642752351352] [4442:201443] CHIP: [IN] 0x08, 0x55, 0x3e, 0x88, 0x87, 0x36, 0x96, 0xba, 
[1642752351352] [4442:201443] CHIP: [IN] 0x0c, 0x36, 0xa5, 0x9b, 0xc2, 0x25, 0xba, 0x11, 
[1642752351352] [4442:201443] CHIP: [IN] 0x4a, 0x7d, 0xc3, 0x35, 0x74, 0x8f, 0x9a, 0x02, 
[1642752351352] [4442:201443] CHIP: [IN] 0xca, 
[1642752351352] [4442:201443] CHIP: [IN] Generated compressed fabric ID
[1642752351352] [4442:201443] CHIP: [IN] 0xff, 0xb5, 0x03, 0x75, 0xbe, 0x2f, 0x4a, 0x94, 
[1642752351352] [4442:201443] CHIP: [DIS] Added new fabric at index: 3, Initialized: 1
[1642752351352] [4442:201443] CHIP: [DIS] Assigned compressed fabric ID: 0xFFB50375BE2F4A94, node ID: 0x000000000001B669
[1642752351352] [4442:201443] CHIP: [CTL] Joined the fabric at index 3. Compressed fabric ID is: 0xFFB50375BE2F4A94
[1642752351352] [4442:201447] CHIP: [DL] _OnPlatformEvent
[1642752351352] [4442:201447] CHIP: [TOO] Sending command to node 0x1
[1642752351353] [4442:201447] CHIP: [DL] Resolved interface id: 8
[... 92 lines cut ...]
[1642752351375] [4442:201447] CHIP: [DMG] 				},
[1642752351375] [4442:201447] CHIP: [DMG] 				
[1642752351375] [4442:201447] CHIP: [DMG] 				StatusIB =
[1642752351375] [4442:201447] CHIP: [DMG] 				{
[1642752351375] [4442:201447] CHIP: [DMG] 					status = 0xc3,
[1642752351375] [4442:201447] CHIP: [DMG] 				},
[1642752351375] [4442:201447] CHIP: [DMG] 				
[1642752351376] [4442:201447] CHIP: [DMG] 			},
[1642752351376] [4442:201447] CHIP: [DMG] 			
[1642752351376] [4442:201447] CHIP: [DMG] 		},
[1642752351376] [4442:201447] CHIP: [DMG] 		
[1642752351376] [4442:201447] CHIP: [DMG] 	],
[1642752351376] [4442:201447] CHIP: [DMG] 	
[1642752351376] [4442:201447] CHIP: [DMG] },
[1642752351376] [4442:201447] CHIP: [DMG] Received Command Response Status for Endpoint=1 Cluster=0x0000_0201 Command=0x0000_0000 Status=0xc3
[1642752351376] [4442:201447] CHIP: [TOO] Default Failure Response: 0xc3
[1642752351376] [4442:201447] CHIP: [DMG] ICR moving to [AwaitingDe]
[1642752351376] [4442:201447] CHIP: [EM] Sending Standalone Ack for MessageCounter:8897135 on exchange 22262i
[1642752351376] [4442:201447] CHIP: [IN] Prepared encrypted message 0x700000619580 to 0x0000000000000001 (1)  of type 0x10 and protocolId (0, 0) on exchange 22262i with MessageCounter:655025.
[1642752351376] [4442:201447] CHIP: [IN] Sending encrypted msg 0x700000619580 with MessageCounter:655025 to 0x0000000000000001 (1) at monotonic time: 29691625 msec
[1642752351376] [4442:201447] CHIP: [EM] Flushed pending ack for MessageCounter:8897135 on exchange 22262i
[1642752351376] [4442:201443] CHIP: [-] ../../zzz_generated/chip-tool/zap-generated/cluster/Commands.h:2502: CHIP Error 0x000000AC: Internal error at ../../examples/chip-tool/commands/common/CHIPCommand.cpp:59
[1642752351376] [4442:201443] CHIP: [TOO] Run command failure: ../../zzz_generated/chip-tool/zap-generated/cluster/Commands.h:2502: CHIP Error 0x000000AC: Internal error
[1642752351378] [4442:201443] CHIP: [SPT] VerifyOrDie failure at ../../src/inet/InetLayer.h:121: sEndPointPool.Allocated() == 0
"

yufengwangca, 2022-02-07:
@chrisbeach59 I saw you implement this logic from git history, could you help to take a look?

chrisbeach59, 2022-03-02:
I have tested this against the linux thermostat app and it works as expected.  
I would need to know more about the DUT configuration to determine if this is a defect of not. 
Please confirm that you have the thermostat cluster present
Make sure that the the heating bit is set in the feature-map attribute.

[1646264494.549878][600252:600252] CHIP:EM: Handling via exchange: 42134r, Delegate: 0x558b0cccbdc0
[1646264494.549905][600252:600252] CHIP:DMG: InvokeRequestMessage =
[1646264494.549911][600252:600252] CHIP:DMG: {
[1646264494.549918][600252:600252] CHIP:DMG: 	suppressResponse = false, 
[1646264494.549927][600252:600252] CHIP:DMG: 	timedRequest = false, 
[1646264494.549934][600252:600252] CHIP:DMG: 	InvokeRequests =
[1646264494.549944][600252:600252] CHIP:DMG: 	[
[1646264494.549950][600252:600252] CHIP:DMG: 		CommandDataIB =
[1646264494.549957][600252:600252] CHIP:DMG: 		{
[1646264494.549964][600252:600252] CHIP:DMG: 			CommandPathIB =
[1646264494.549972][600252:600252] CHIP:DMG: 			{
[1646264494.549979][600252:600252] CHIP:DMG: 				EndpointId = 0x1,
[1646264494.549987][600252:600252] CHIP:DMG: 				ClusterId = 0x201,
[1646264494.549995][600252:600252] CHIP:DMG: 				CommandId = 0x0,
[1646264494.550005][600252:600252] CHIP:DMG: 			},
[1646264494.550015][600252:600252] CHIP:DMG: 			
[1646264494.550022][600252:600252] CHIP:DMG: 			CommandData = 
[1646264494.550030][600252:600252] CHIP:DMG: 			{
[1646264494.550039][600252:600252] CHIP:DMG: 				0x0 = 0, 
[1646264494.550051][600252:600252] CHIP:DMG: 				0x1 = 30, 
[1646264494.550060][600252:600252] CHIP:DMG: 			},
[1646264494.550067][600252:600252] CHIP:DMG: 		},
[1646264494.550077][600252:600252] CHIP:DMG: 		
[1646264494.550084][600252:600252] CHIP:DMG: 	],
[1646264494.550095][600252:600252] CHIP:DMG: