project-chip/connectedhomeip #13880: Matter Crash on iOS

woody-apple, 2022-01-24:
#### Problem
Pretty common crash here:

```
Thread 4 Crashed ↩::   Dispatch queue: com.zigbee.chip.framework.controller.workqueue
0   libsystem_kernel.dylib        	       0x1a5939270 __pthread_kill + 8
1   libsystem_pthread.dylib       	       0x1a59cd474 pthread_kill + 272
2   libsystem_c.dylib             	       0x1a589f474 abort + 164
3   libsystem_malloc.dylib        	       0x1a5999a5c malloc_vreport + 552
4   libsystem_malloc.dylib        	       0x1a5999c34 malloc_report + 64
5   libsystem_malloc.dylib        	       0x1a59916a8 free + 300
6   CHIP                          	       0x11200a55e chip::Platform::MemoryFree(void*) + 132
7   CHIP                          	       0x11200d26e void chip::Platform::Delete<chip::Messaging::ExchangeMessageDispatch>(chip::Messaging::ExchangeMessageDispatch*) + 72
8   CHIP                          	       0x11200d21a chip::DeleteDeletor<chip::Messaging::ExchangeMessageDispatch>::Release(chip::Messaging::ExchangeMessageDispatch*) + 28
9   CHIP                          	       0x11200c07a chip::ReferenceCounted<chip::Messaging::ExchangeMessageDispatch, chip::DeleteDeletor<chip::Messaging::ExchangeMessageDispatch>, 1>::Release() + 80
10  CHIP                          	       0x11200bfe2 chip::Messaging::ExchangeContext::~ExchangeContext() + 312
11  CHIP                          	       0x11200c0aa chip::Messaging::ExchangeContext::~ExchangeContext() + 32
12  CHIP                          	       0x11200cf3e chip::BitMapObjectPool<chip::Messaging::ExchangeContext, 8ul,>::ReleaseObject(chip::Messaging::ExchangeContext*) + 56
13  CHIP                          	       0x11200babe chip::Messaging::ExchangeManager::ReleaseContext(chip::Messaging::ExchangeContext*) + 40
14  CHIP                          	       0x11200ba8a chip::Messaging::ExchangeContextDeletor::Release(chip::Messaging::ExchangeContext*) + 36
15  CHIP                          	       0x11200b99e chip::ReferenceCounted<chip::Messaging::ExchangeContext, chip::Messaging::ExchangeContextDeletor, 1>::Release() + 80
16  CHIP                          	       0x11200d1ee chip::ReferenceCountedHandle<chip::Messaging::ExchangeContext>::~ReferenceCountedHandle() + 40
17  CHIP                          	       0x11200b7b6 chip::ReferenceCountedHandle<chip::Messaging::ExchangeContext>::~ReferenceCountedHandle() + 32
18  CHIP                          	       0x112011296 chip::Messaging::ReliableMessageMgr::RetransTableEntry::~RetransTableEntry() + 64
19  CHIP                          	       0x1120112c6 chip::Messaging::ReliableMessageMgr::RetransTableEntry::~RetransTableEntry() + 32
20  CHIP                          	       0x1120120a2 chip::BitMapObjectPool<chip::Messaging::ReliableMessageMgr::RetransTableEntry, 8ul,>::ReleaseObject(chip::Messaging::ReliableMessageMgr::RetransTableEntry*) + 56
21  CHIP                          	       0x112012656 auto chip::Messaging::ReliableMessageMgr::Shutdown()::$_0::operator()<chip::Messaging::ReliableMessageMgr::RetransTableEntry>(chip::Messaging::ReliableMessageMgr::RetransTableEntry*) const + 44
22  CHIP                          	       0x1120125c2 chip::internal::LambdaProxy<chip::Messaging::ReliableMessageMgr::RetransTableEntry, chip::Messaging::ReliableMessageMgr::Shutdown()::$_0>::Call(void*, void*) + 36
23  CHIP                          	       0x11200616a chip::internal::StaticAllocatorBitmap::ForEachActiveObjectInner(void*, chip::Loop*)(void*, void*)) + 280
24  CHIP                          	       0x112011536 chip::Loop chip::BitMapObjectPool<chip::Messaging::ReliableMessageMgr::RetransTableEntry, 8ul,>::ForEachActiveObject<chip::Messaging::ReliableMessageMgr::Shutdown()::$_0>(chip::Messaging::ReliableMessageMgr::Shutdown()::$_0&&) + 88
25  CHIP                          	       0x11201146a chip::Messaging::ReliableMessageMgr::Shutdown() + 52
26  CHIP                          	       0x11200e75e chip::Messaging::ExchangeManager::Shutdown() + 40
27  CHIP                          	       0x111fd2a4e chip::Controller::DeviceControllerSystemState::Shutdown() + 308
28  CHIP                          	       0x111fc037a chip::Controller::DeviceControllerSystemState::Release() + 100
29  CHIP                          	       0x111fc02b2 chip::Controller::DeviceController::Shutdown() + 188
30  CHIP                          	       0x111fc411a chip::Controller::DeviceCommissioner::Shutdown() + 120
31  CHIP                          	       0x11143cc02 __32-[XXXXXXX shutdown]_block_invoke + 260
32  libdispatch.dylib             	       0x1a57a7134 _dispatch_call_block_and_release + 32
33  libdispatch.dylib             	       0x1a57a8cf4 _dispatch_client_callout + 20
34  libdispatch.dylib             	       0x1a57b0408 _dispatch_lane_serial_drain + 668
35  libdispatch.dylib             	       0x1a57b0f64 _dispatch_lane_invoke + 392
36  libdispatch.dylib             	       0x1a57bb7a8 _dispatch_workloop_worker_thread + 648
37  libsystem_pthread.dylib       	       0x1a59ce2d4 _pthread_wqthread + 288
38  libsystem_pthread.dylib       	       0x1a59d5618 start_wqthread + 8
Thread 5 
0   libsystem_pthread.dylib       	       0x1a59d5610 start_wqthread + 0
Thread 6  Dispatch queue: XXXXXX
0   libsystem_kernel.dylib        	       0x1a593b0b8 kevent_id + 8
1   libdispatch.dylib             	       0x1a57cb314 _dispatch_kq_poll + 228
2   libdispatch.dylib             	       0x1a57cbd38 _dispatch_event_loop_wait_for_ownership + 444
3   libdispatch.dylib             	       0x1a57b8200 __DISPATCH_WAIT_FOR_QUEUE__ + 328
4   libdispatch.dylib             	       0x1a57b7dd4 _dispatch_sync_f_slow + 144
5   CHIP                          	       0x1120a4016 chip::DeviceLayer::PlatformManagerImpl::_StopEventLoopTask() + 176
6   CHIP                          	       0x11143cce2 chip::DeviceLayer::PlatformManager::StopEventLoopTask() + 28
7   CHIP                          	       0x11143cab2 -[XXXXXXX shutdown] + 176
8   CHIPPlugin                    	       0x104f12092 -[XXXXXXX _restartChipStack] + 216
9   CHIPPlugin                    	       0x104f11f9a __46-[XXXXXXX restartChipStack]_block_invoke + 40
10  libdispatch.dylib             	       0x1a57a7134 _dispatch_call_block_and_release + 32
11  libdispatch.dylib             	       0x1a57a8cf4 _dispatch_client_callout + 20
12  libdispatch.dylib             	       0x1a57b0408 _dispatch_lane_serial_drain + 668
13  libdispatch.dylib             	       0x1a57b0f64 _dispatch_lane_invoke + 392
14  libdispatch.dylib             	       0x1a57bb7a8 _dispatch_workloop_worker_thread + 648
15  libsystem_pthread.dylib       	       0x1a59ce2d4 _pthread_wqthread + 288
16  libsystem_pthread.dylib       	       0x1a59d5618 start_wqthread + 8
```

 #### Proposed Solution
<suggested fix, suggested enhancement>

bzbarsky-apple, 2022-01-25:
@woody-apple Using which SHA?  Does it have PR #13448 in it already?

bzbarsky-apple, 2022-01-25:
Looks like those stacks are from builds that do not have #13448 in them.  I believe that PR should resolve this problem, so closing.  If this keeps happening after that PR is in place, we should re-evaluate.

closed 2022-01-25 by bzbarsky-apple