project-chip/connectedhomeip #13899: Cannot commission m5stack using CHIP Tool on iOS 15.2

m-cheung, 2022-01-25:
I'm not sure if anything has changed recently, but it seems like I can no longer commission my m5stack using the CHIP Tool. I've been poking around and it seems like the device is not being detected by `CBCentralManager` within `BleConnectionDelegateImpl`.

I've been poking around using other tools (e.g. Lightblue and nRF) and it seems like they do not see the m5stack advertising the CHIP service either on iOS 15.2.1.

I asked a colleague with an iOS 12.5.5 and iOS 15.1 device to check Lightblue and nRF with their m5stack. They mentioned that they see the CHIP service being advertised using those devices. They upgraded their iOS 15.1 device to iOS 15.2.1 device and they stopped seeing the service being advertised within Lightblue and nRF.

Is there something we are overlooking here?

(tagging some of the Apple team for visibility @carol-apple , @vivien-apple @pan-apple @bzbarsky-apple)

bzbarsky-apple, 2022-01-25:
@sagar-apple

woody-apple, 2022-01-26:
@woody-apple Can you please reach out to me over slack? or via @andy31415 ? Thanks!

closed 2022-01-26 by woody-apple