project-chip/connectedhomeip #13919: The ClusterInfo may support a group id, the iterator should be able to call group data provider to expand the group id.

woody-apple, 2022-01-25:
false

---

https://github.com/project-chip/connectedhomeip/blob/f0f5de0d79ae1c51d7fddbf70900ebac0acd899a/src/app/AttributePathExpandIterator.h#L67-L72

---

###### This issue was generated by [todo-issue](https://github.com/DerJuulsn/todo-issue) based on a `TODO` comment in f0f5de0d79ae1c51d7fddbf70900ebac0acd899a.

woody-apple, 2022-02-08:
@rcasallas-silabs ?

mrjerryjohns, 2022-02-25:
It looks like `WriteHandler` already got updated to expand group paths, but that logic wasn't added to `AttributePathExpandIterator`.

The cleaner option here might be to move it to the expand iterator as the final home for that logic, but this is just purely a clean-up exercise.

Punting out of 1.0.

closed 2022-09-22 by stale[bot]