project-chip/connectedhomeip #13928: The code currently has two sources of truths for fabrics, the fabricInfo table + the attributes. There should only be one,

woody-apple, 2022-01-25:
the attributes list. Currently the attributes are not persisted so we are keeping the fabric table to have the fabrics/admrins be persisted. Once attributes are persisted, there should only be one sorce of truth, the attributes list and only that should be modifed to perosst/read/write fabrics. TODO: Once attributes are persisted, implement reading/writing/manipulation fabrics around that and remove fabricTable logic.

---

https://github.com/project-chip/connectedhomeip/blob/f0f5de0d79ae1c51d7fddbf70900ebac0acd899a/src/app/clusters/operational-credentials-server/operational-credentials-server.cpp#L248-L253

---

###### This issue was generated by [todo-issue](https://github.com/DerJuulsn/todo-issue) based on a `TODO` comment in f0f5de0d79ae1c51d7fddbf70900ebac0acd899a.

holbrookt, 2022-01-31:
@pan-apple it looks like you were the last to edit this comment. Is this a concern for 1.0?

bzbarsky-apple, 2022-06-13:
This is fixed: the fabric table is the source of truth now, and the attributes come from an AttributeAccessInterface that reads it.

closed 2022-06-13 by bzbarsky-apple