project-chip/connectedhomeip #13929: TE7.5 [TC-MF-1.3] and [TC-MF-1.4] pairing fail on 2nd controller

wilbur-se, 2022-01-25:
branch: TE7.5
device: wifi smart plug

problem:
    TC-MF-1.3:
        1.open-commissioning-window always fail after commissioned, but will success after reboot device
        2.when open-commissioning-window success, pairing manualcode on the 2nd controller  always fail
   TC-MF-1.4:
        pairing onnetwork on the 2nd controller always fail

[TC-MF-1.3_1.txt](https://github.com/project-chip/connectedhomeip/files/7931335/TC-MF-1.3_1.txt)
[TC-MF-1.3_2.txt](https://github.com/project-chip/connectedhomeip/files/7931336/TC-MF-1.3_2.txt)
[TC-MF-1.3_2-fail.txt](https://github.com/project-chip/connectedhomeip/files/7931339/TC-MF-1.3_2-fail.txt)
[TC-MF-1.3_3.txt](https://github.com/project-chip/connectedhomeip/files/7931340/TC-MF-1.3_3.txt)
[TC-MF-1.4_1.txt](https://github.com/project-chip/connectedhomeip/files/7931343/TC-MF-1.4_1.txt)
[TC-MF-1.4_2.txt](https://github.com/project-chip/connectedhomeip/files/7931345/TC-MF-1.4_2.txt)
[TC-MF-1.4_3.txt](https://github.com/project-chip/connectedhomeip/files/7931347/TC-MF-1.4_3.txt)

bzbarsky-apple, 2022-03-15:
In the failure case, we send a read request and never seem to get a response, right?

wwx0715, 2022-03-16:
For TC-MF-1.3, we send 'open-commissioning-window' request, for TC-MF-1.4, we send 'pairing onnetwork'. No any response for all the request, and get a TIMEOUT at last

wilbur-se, 2022-03-16:
For TC-MF-1.3, we send 'open-commissioning-window' request, for TC-MF-1.4, we send 'pairing onnetwork'. No any response for all the request, and get a TIMEOUT at last

naveenkommareddi, 2022-05-17:
Dup of #18084. Fails on Thread transport.