project-chip/connectedhomeip #13938: Will need to actually figure out how to test for this!

woody-apple, 2022-01-25:
false

---

https://github.com/project-chip/connectedhomeip/blob/f0f5de0d79ae1c51d7fddbf70900ebac0acd899a/src/app/util/util.cpp#L470-L475

---

###### This issue was generated by [todo-issue](https://github.com/DerJuulsn/todo-issue) based on a `TODO` comment in f0f5de0d79ae1c51d7fddbf70900ebac0acd899a.

turon, 2022-01-31:
Testing multicast will be important once group messaging is complete.
Tagged task as `group` and removed `v1_triage_split_4 `.

jepenven-silabs, 2022-02-23:
Closing for Multiple reasons : 
1. Comment not present anymore in the SDK (assumes resolved)
2. Group messaging, including multicast can be tested on Linux, Darwin and Any other embedded device, including Thread devices using the `TestGroupMessaging.yaml` which issue multicast commands after configuring the right groups. and group keys

closed 2022-02-23 by jepenven-silabs