project-chip/connectedhomeip #13945: Support opening commissioning window simultaneously on multiple devices

woody-apple, 2022-01-25:
false

---

https://github.com/project-chip/connectedhomeip/blob/f0f5de0d79ae1c51d7fddbf70900ebac0acd899a/src/controller/CHIPDeviceController.h#L404-L409

---

###### This issue was generated by [todo-issue](https://github.com/DerJuulsn/todo-issue) based on a `TODO` comment in f0f5de0d79ae1c51d7fddbf70900ebac0acd899a.

woody-apple, 2022-02-08:
Could be possible to just support 1 commissioning at a time for the v1 SDK...

cecille, 2022-02-25:
I don't think this is necessary for 1.0 - platforms that really need to open commissioning windows to multiple devices simultaneously can do so by calling the appropriate cluster command. We don't currently support commissioning multiple devices simultaneously from the same DeviceController anyway, so opening multiple windows simultaneously from one controller is not a feature that is likely to get much use.

closed 2022-09-22 by stale[bot]