project-chip/connectedhomeip #13965: Drop API level to 21 after getifaddrs issue fixed

woody-apple, 2022-01-25:
false

---

https://github.com/project-chip/connectedhomeip/blob/f0f5de0d79ae1c51d7fddbf70900ebac0acd899a/build/toolchain/android/BUILD.gn#L43-L48

---

###### This issue was generated by [todo-issue](https://github.com/DerJuulsn/todo-issue) based on a `TODO` comment in f0f5de0d79ae1c51d7fddbf70900ebac0acd899a.

andy31415, 2022-01-31:
Due to RIO requirements for IPv6, minimum android for chipt is 8.0, which is api level 26.
Need followup with java team if this is really required for other reasons.

andy31415, 2022-02-15:
I had no complaints so far about java integration and integration seems to generally require api level 26 or custom code. I don't think this is a high priority for V1 at this point given the rest of the work that is still outstanding.

closed 2022-09-22 by stale[bot]