project-chip/connectedhomeip #13973: enable CommissionedListCommand once DNS Cache is implemented

woody-apple, 2022-01-25:
make_unique&lt;CommissionedListCommand&gt;(),

---

https://github.com/project-chip/connectedhomeip/blob/f0f5de0d79ae1c51d7fddbf70900ebac0acd899a/examples/chip-tool/commands/pairing/Commands.h#L199-L204

---

###### This issue was generated by [todo-issue](https://github.com/DerJuulsn/todo-issue) based on a `TODO` comment in f0f5de0d79ae1c51d7fddbf70900ebac0acd899a.

turon, 2022-02-01:
All commands specified for commissioning are required for v1.
Removing tag `v1_triage_split_4`

cecille, 2022-02-25:
@woody-apple - can you specify what the spec issue is here? This looks like a chip-tool feature request to me. I'm possibly misunderstanding what this issue is for.

bzbarsky-apple, 2022-05-05:
This is in fact a chip-tool feature, and not needed for 1.0

closed 2023-06-10 by stale[bot]