project-chip/connectedhomeip #13984: handle multiple transfers by tracking updateTokens

woody-apple, 2022-01-25:
false

---

https://github.com/project-chip/connectedhomeip/blob/f0f5de0d79ae1c51d7fddbf70900ebac0acd899a/examples/ota-provider-app/ota-provider-common/OTAProviderExample.cpp#L185-L190

---

###### This issue was generated by [todo-issue](https://github.com/DerJuulsn/todo-issue) based on a `TODO` comment in f0f5de0d79ae1c51d7fddbf70900ebac0acd899a.

woody-apple, 2022-02-08:
@carol-apple , post v1?

carol-apple, 2022-02-17:
Yes, this should be post v1. Currently, we're only supporting one active transfer at a time.

carol-apple, 2022-06-01:
Should multiple transfers only be applicable to the provider? Does it make sense for a requestor to maintain multiple transfers at a time? This does not seem to be likely needed for a requestor. The current spec seems to only mention that the provider uses the `updateToken` to maintain multiple transfers and that requestor just has to send it to provider for the entirety of the OTA process.

Assignee of this ticket should follow up with spec side to see if the intention of "multiple transfers" is not applicable to the requestor.

closed 2022-12-11 by stale[bot]