project-chip/connectedhomeip #14045: Currently, DeviceNetworkProvisioningDelegateImpl assumes that ssid and credentials are null terminated strings,

woody-apple, 2022-01-25:
which is not correct, this should be changed once we have better method for commissioning wifi networks.

---

https://github.com/project-chip/connectedhomeip/blob/f0f5de0d79ae1c51d7fddbf70900ebac0acd899a/src/app/clusters/network-commissioning-old/network-commissioning-old.cpp#L241-L246

---

###### This issue was generated by [todo-issue](https://github.com/DerJuulsn/todo-issue) based on a `TODO` comment in f0f5de0d79ae1c51d7fddbf70900ebac0acd899a.

mrjerryjohns, 2022-01-31:
Looks critical, since just a reinterpret_cast is not going to cut it for strings send in commands.

woody-apple, 2022-02-08:
I believe this is dead code that needs to be removed. Moving post v1.

cecille, 2022-02-25:
This todo appears in the old network commissioning framework and this file has now been removed. Closing as obsolete.

closed 2022-02-25 by cecille