project-chip/connectedhomeip #14049: Needs implementation.

woody-apple, 2022-01-25:
false

---

https://github.com/project-chip/connectedhomeip/blob/f0f5de0d79ae1c51d7fddbf70900ebac0acd899a/src/app/clusters/power-source-server/power-source-server.cpp#L55-L60

---

###### This issue was generated by [todo-issue](https://github.com/DerJuulsn/todo-issue) based on a `TODO` comment in f0f5de0d79ae1c51d7fddbf70900ebac0acd899a.

turon, 2022-02-01:
PowerSource cluster must comply with spec for v1.
Removing tag `v1_triage_split_4`

cjandhyala, 2022-05-27:
@fessehaeve @woody-apple , We do have a test case to read that Attribute. But per spec this attribute is optional, So we are not verifying for optional attributes for V1.0. Ref spec ref

[image]

andy31415, 2022-05-30:
Moved post 1.0 since this functionality is optional

closed 2022-12-05 by stale[bot]