project-chip/connectedhomeip #14063: This does not play nicely with wildcard reads.  Need to fix ZAP to

woody-apple, 2022-01-25:
put it in the metadata, or fix wildcard expansion to add it.

---

https://github.com/project-chip/connectedhomeip/blob/f0f5de0d79ae1c51d7fddbf70900ebac0acd899a/src/app/util/ember-compatibility-functions.cpp#L307-L312

---

###### This issue was generated by [todo-issue](https://github.com/DerJuulsn/todo-issue) based on a `TODO` comment in f0f5de0d79ae1c51d7fddbf70900ebac0acd899a.

holbrookt, 2022-01-31:
Hi Boris, it looks like you added this TODO. Is this needed for 1.0? If not please remove the label. Thanks!

bzbarsky-apple, 2022-02-01:
This is a spec-compliance issue, so yes, needed.  That said, we should be able to fix this without too much trouble.  But we should merge https://github.com/project-chip/connectedhomeip/pull/14299 first and then fix all three attributes together.

closed 2022-02-15 by bzbarsky-apple