project-chip/connectedhomeip #14075: HWADR not supported yet

woody-apple, 2022-01-25:
false

---

https://github.com/project-chip/connectedhomeip/blob/f0f5de0d79ae1c51d7fddbf70900ebac0acd899a/src/app/zap-templates/zcl/data-model/chip/general-diagnostics-cluster.xml#L74-L79

---

###### This issue was generated by [todo-issue](https://github.com/DerJuulsn/todo-issue) based on a `TODO` comment in f0f5de0d79ae1c51d7fddbf70900ebac0acd899a.

bzbarsky-apple, 2022-02-10:
@yufengwangca

bzbarsky-apple, 2022-05-05:
Might not be worth fixing this for 1.0.  The actual code and spec-observable behavior is correct.  So the only change to be made here would be to use `hwadr` in the XML and then plumb though all the codegen bits to make the code come out exactly the same in the end.

Unless we want a non-ByteSpan representation of this on the C++ side, of course, but it's not clear that we want that.

d3vi1, 2026-10-24:
From my understanding of the SPEC, all mac addresses (length 6) require prepending with 00h 00h in order to be compliant, however, all Project-CHIP based implementations serve the mac addresses without the prepend, thus creating bugs for non project-CHIP based implementations that attempt to be compliant with the SPEC. We should have some sort of guidance on this.