project-chip/connectedhomeip #14076: Add feature map once it is available

woody-apple, 2022-01-25:
false

---

https://github.com/project-chip/connectedhomeip/blob/f0f5de0d79ae1c51d7fddbf70900ebac0acd899a/src/app/zap-templates/zcl/data-model/chip/keypad-input-cluster.xml#L26-L31

---

###### This issue was generated by [todo-issue](https://github.com/DerJuulsn/todo-issue) based on a `TODO` comment in f0f5de0d79ae1c51d7fddbf70900ebac0acd899a.

andy31415, 2022-01-31:
Spec KeypadInput defines the following features:

```
105 === Features
106 This cluster SHALL support the FeatureMap bitmap attribute as defined below.
107
108 [options="header",valign="middle"]
109 |===
110 s| Bit | Code | Feature             | Description
111 s| 0   | NV   | Navigation KeyCodes | Supports UP, DOWN, LEFT, RIGHT, SELECT, BACK, EXIT, MENU
112 s| 1   | LK   | Location Keys       | Supports CEC keys 0x0A (Settings) and 0x09 (Home)
113 s| 2   | NK   | Number Keys         | Supports numeric input 0..9
114 |===
```

bzbarsky-apple, 2022-05-05:
The XML has:
```
  <bitmap name="KeypadInputFeature" type="BITMAP32">
    <cluster code="0x0509"/>
    <field name="NavigationKeyCodes" mask="0x1"/>
    <field name="LocationKeys" mask="0x2"/>
    <field name="NumberKeys" mask="0x4"/>
  </bitmap>
```
and pas that this is about setting the right value in the ZAP config, right?

That TODO comment seems to be gone.

@chrisdecenzo what's left to do here?

cjandhyala, 2022-05-27:
@woody-apple @bzbarsky-apple could this be dup to [#16725](https://github.com/project-chip/connectedhomeip/issues/16725)

bzbarsky-apple, 2022-05-28:
Duplicate of https://github.com/project-chip/connectedhomeip/issues/16725

closed 2022-05-28 by bzbarsky-apple