project-chip/connectedhomeip #14078: Do events work?

woody-apple, 2022-01-25:
false

---

https://github.com/project-chip/connectedhomeip/blob/f0f5de0d79ae1c51d7fddbf70900ebac0acd899a/src/app/zap-templates/zcl/data-model/chip/power-source-cluster.xml#L66-L71

---

###### This issue was generated by [todo-issue](https://github.com/DerJuulsn/todo-issue) based on a `TODO` comment in f0f5de0d79ae1c51d7fddbf70900ebac0acd899a.

bzbarsky-apple, 2022-05-05:
Events generally work.  Nothing right now emits this specific event, as far as I can tell....

cjandhyala, 2022-05-27:
@woody-apple Events in the Power source config cluster are optional per latest spec, hence we are not testing for v1.0
[image]

andy31415, 2022-05-30:
Moving out of v1.0 functionality that is optional

closed 2023-10-15 by stale[bot]