project-chip/connectedhomeip #14081: check that there is no commissioning in progress currently.

woody-apple, 2022-01-25:
false

---

https://github.com/project-chip/connectedhomeip/blob/f0f5de0d79ae1c51d7fddbf70900ebac0acd899a/src/controller/AutoCommissioner.cpp#L195-L200

---

###### This issue was generated by [todo-issue](https://github.com/DerJuulsn/todo-issue) based on a `TODO` comment in f0f5de0d79ae1c51d7fddbf70900ebac0acd899a.

mrjerryjohns, 2022-01-31:
Rationale for keeping it v1: Ensuring we don't start commissioning when one is in progress seems like good house-keeping to do to prevent worse failures from cropping up later.

woody-apple, 2022-02-08:
@cecille can you take a look at this, also - note this one:
https://github.com/project-chip/connectedhomeip/issues/13945

PR #15870 (2022-03-04): Remove stale todo
#### Problem
Commissioning state is already checked by the caller. We don't need to double track it in the auto commissioner. We should just take the DeviceCommissioner's word that we need to start commissioning. Other implementations may track if they wish, but this example impl doesn't need it and tracking the state in more than once place can lead to bugs.
Fixes: #14081

#### Change overview
Remove todo. We don't want to do it.

#### Testing
Comment removal only.
closed 2022-03-04 by woody-apple