project-chip/connectedhomeip #14083: Get real endpoint

woody-apple, 2022-01-25:
false

---

https://github.com/project-chip/connectedhomeip/blob/f0f5de0d79ae1c51d7fddbf70900ebac0acd899a/src/controller/AutoCommissioner.cpp#L312-L317

---

###### This issue was generated by [todo-issue](https://github.com/DerJuulsn/todo-issue) based on a `TODO` comment in f0f5de0d79ae1c51d7fddbf70900ebac0acd899a.

turon, 2022-02-01:
`PerformCommissioningStep` needs to act on expected endpoint and not only endpoint 0.
Assuming this needs to be fixed for v1 for spec compliance.
Not sure which use cases aren't supported by the current "default to endpoint 0" assumption.
Removing tag `v1_triage_split_4`.

bzbarsky-apple, 2022-06-13:
This is fixed.  We're now doing `GetEndpoint(nextStage)` at the relevant callsite.

closed 2022-06-13 by bzbarsky-apple