project-chip/connectedhomeip #14146: Aborted(core dumped)  occured frequently duiring execution of test cases and commissioning process

nanmu2018, 2022-01-25:
#### Problem
@jmartinez-silabs
**Problem：**
1. commissioning process
[1643094311.815623][3879:3879] CHIP:SPT: VerifyOrDie failure at ../../examples/chip-tool/third_party/connectedhomeip/src/inet/InetLayer.h:121: sEndPointPool.Allocated() == 0
Aborted
2.execution of test cases
[1643081773.668925][4000:4000] CHIP:SPT: VerifyOrDie failure at ../../examples/chip-tool/third_party/connectedhomeip/src/inet/InetLayer.h:121: sEndPointPool.Allocated() == 0
Aborted (core dumped)

**platform:**
Testing image on RasPi：v1.7.img
Connectedhomeip commitID : b0af6ba42b6ce494748f18ecf3e054a400b8f45b
OTBR commitID：72bb3d45684f837e67e7f1ec20a9a8e3ac4a4419
lighting-app run on the EFR32MG24 board base on commitID : b0af6ba42b6ce494748f18ecf3e054a400b8f45b

**commissioning logs attached here**
[commissioning.txt](https://github.com/project-chip/connectedhomeip/files/7932021/Aborted.during.commissioning.txt)
[lighting-Aborted during commissioning.txt](https://github.com/project-chip/connectedhomeip/files/7932023/lighting-Aborted.during.commissioning.txt)
[otbr_log.txt](https://github.com/project-chip/connectedhomeip/files/7932024/otbr_log.txt)

**execution of test cases logs attached here**
Failing comman：sudo ./chip-tool descriptor read device-list 12344321 1
[execution of test cases.txt](https://github.com/project-chip/connectedhomeip/files/7932043/execution.of.test.cases.txt)
[lighting-Aborted duiring execution of test cases.txt](https://github.com/project-chip/connectedhomeip/files/7932045/lighting-Aborted.duiring.execution.of.test.cases.txt)
[otbr_log.txt](https://github.com/project-chip/connectedhomeip/files/7932046/otbr_log.txt)

jmartinez-silabs, 2022-01-25:
What is your chip-tool and otbr setup ? Are they both on the same raspi or separate?
What is the IPV6 address of your raspi running chip-tool?

The lighting app does not receive the CASE SIGMA1 message from chip-tool. I don't see on the otbr log any packet sent to 2001:db8::e784:2d6a:dc30:8a36 (The Global IPV6 of your lighting app that the CHIP-tool is sending to)

Therefore, I think there is a routing issue on setup.

nanmu2018, 2022-01-27:
@jmartinez-silabs
My chip-tool and otbr are on the same raspi.

The way to setup my otbr manually:
- git clone https://github.com/openthread/ot-br-posix 
- cd ot-br-posix 
- git checkout 72bb3d45684f837e67e7f1ec20a9a8e3ac4a4419
-  ./script/bootstrap 
- sudo BACKBONE_ROUTER=1 INFRA_IF_NAME=eth0 ./script/setup

The way to setup chip-tool:
- ./scripts/examples/gn_build_example.sh examples/chip-tool out/standalone 

I restarted my raspi and completed the commissioning successfully, but I still  get Abort failures during commissioning frequently.
It fails here:
Ipv6 of lighting device:   2001:DB8::6157:481B:8573:1B1A
Ipv6 of otbr :  2001:db8:0:0:3f3d:7dde:6d01:f667
[otbr_log.txt](https://github.com/project-chip/connectedhomeip/files/7948114/otbr_log.txt)
[light_log.txt](https://github.com/project-chip/connectedhomeip/files/7948116/light_log.txt)
[commissioning.txt](https://github.com/project-chip/connectedhomeip/files/7948120/commissioning.txt)

andy31415, 2022-02-07:
I believe the root cause is `[1643081773.661250][4000:4005] CHIP:SC: CASESession timed out while waiting for a response from the peer. Current state was 1`

We do not handle non-happy paths and we end with a core dump on cleanup, however by that time all commands already failed.

cecille, 2022-03-29:
@Minan-Tuyasmart - is this still happening for you?

closed 2022-10-03 by stale[bot]