project-chip/connectedhomeip #14159: haveIPv6Conn */

woody-apple, 2022-01-25:
false

---

https://github.com/project-chip/connectedhomeip/blob/e169fcf99510c98ea16504e77c7fd8bf86f6b806/src/platform/EFR32/ConnectivityManagerImpl_WIFI.cpp#L406-L411

---

###### This issue was generated by [todo-issue](https://github.com/DerJuulsn/todo-issue) based on a `TODO` comment in e169fcf99510c98ea16504e77c7fd8bf86f6b806.

holbrookt, 2022-01-31:
Hi Narayan, it looks like this TODO came from https://github.com/project-chip/connectedhomeip/pull/13193 - is this something that needs to be fixed for 1.0?

selissia, 2022-02-01:
@mrfinesse99 , could you please take a look?

mrfinesse99, 2022-02-01:
That TODO was there as a place holder for RS911x - However it is no longer needed as the abstraction that interfaces to lower level of WiFi code handles this correctly. So that comment of TODO can be safely ignored for 1.0

closed 2022-02-03 by woody-apple