project-chip/connectedhomeip #14225: [ TC-DIAG-TH_NW-1.2]- could not able to verify "802.15.4" channel number,Need yaml support for this

kowsisoundhar12, 2022-01-25:
Summary:
As of now not able to verify "802.15.4" channel number in channel attribute ,need yaml support for this 
Test Plan Screenshot:
[image]
Spec-Screenshot :
[image]

vivien-apple, 2022-01-27:
@kowsisoundhar12 Sorry I'm not sure to understand the issue. I assume `Channel` contains a number which is the 802.15.4 channel number so likely some value between 11 and 26 ?
Why are you not able to verify that ?

kowsisoundhar12, 2022-02-02:
@vivien-apple i have tried the various value inbetween 11 and 26 for channel attribute  but getting value mismatch issue
[image]
[image]
[err-logs.zip](https://github.com/project-chip/connectedhomeip/files/7983761/err-logs.zip)

vivien-apple, 2022-02-02:
@kowsisoundhar12 
So per my understanding what you want is to have a `configuredChannel` field in the config section that defaults to `0` for CI.
Then instead of `12` (or any other number you need to use 'configuredChannel`.

This setting should let you pass `--configured-channel` as a command line argument to the command depending on the configured channel of the thread interface.

raju-apple, 2022-02-23:
Script modification needed

kowsisoundhar12, 2022-03-22:
@raju-apple @vivien-apple i have updated the script based on above suggestion  and tested ,its working fine 
please refer attached screenshot and logs 
[image]
[execution-log-march22.zip](https://github.com/project-chip/connectedhomeip/files/8323307/execution-log-march22.zip)

closed 2022-03-22 by kowsisoundhar12