project-chip/connectedhomeip #14228: Identify command in chip-tool fails when IdentifyTime attribute value is of type uint16_t

PSONALl, 2022-01-25:
#### Problem
IdentifyTime attribute is of type uint16_t but for any value greater than 254, Identify Command returns failure on chip-tool

 * expected behavior - Identify command must return success
 * steps to reproduce - commission the device and send Identify command with IdentifyTime value greater than 254
 * system configuration 
            - device : esp32
 *  logs 
            - chip-tool logs: [chip-tool.txt](https://github.com/project-chip/connectedhomeip/files/7933023/chip-tool.txt)
            - device logs: [esp32.txt](https://github.com/project-chip/connectedhomeip/files/7933027/esp32.txt)

bzbarsky-apple, 2022-05-19:
@PSONALl Are you still seeing this?  It seems to work for me on tip, at least against a MacOS server.....

shubhamdp, 2022-06-13:
Tested on lighting-app/esp32 and works fine for me on commit:c1e1d018820c0fdd51f124f607b9207f31fae2ff

sstruchtrup, 2022-06-14:
Works for me as well, tested against the lighting-app example (commit 5148b5d00154f272e0fd250140dd94fbd33c958e) with chip-tool built from the TE9 tag.

closed 2022-06-16 by PSONALl