project-chip/connectedhomeip #14249: What about the cluster-specific part of the status?

woody-apple, 2022-01-25:
false

---

https://github.com/project-chip/connectedhomeip/blob/d0cbf834ccb8d466970de2456e61b5f6412d9bdc/src/darwin/Framework/CHIP/CHIPError.mm#L39-L44

---

###### This issue was generated by [todo-issue](https://github.com/DerJuulsn/todo-issue) based on a `TODO` comment in d0cbf834ccb8d466970de2456e61b5f6412d9bdc.

woody-apple, 2022-02-09:
Let's just have a sub NSError here

bzbarsky-apple, 2022-03-11:
What we current have implemented is:
```
/**
 * MatterInteractionErrorDomain contains errors that represent a Matter
 * StatusIB error.  These represent errors reported by the other side of a
 * Matter interaction.
 *
 * When the code is MatterInteractionErrorCodeFailure the userInfo may have a
 * key named @"clusterStatus" whose value is the cluster-specific status that
 * was reported.  This key will be absent if there was no cluster-specific
 * status.
 */
```

We could do the sub NSError instead if that's better?

woody-apple, 2022-04-22:
Works fine as is :)

woody-apple, 2022-04-22:
@bzbarsky-apple

closed 2022-04-22 by bzbarsky-apple