project-chip/connectedhomeip #14260: Timeouts in Darwin seem to be capped at a max of 3 secs , some of the TC require upto 15 secs in some cases

raju-apple, 2022-01-25:
Is this by design ?

Probably the issue could be related to this : 
https://github.com/project-chip/connectedhomeip/blob/master/src/darwin/Framework/CHIP/templates/clusters-tests.zapt#L22

Can this be fixed ? Blocking us from adding a few CC test cases.

woody-apple, 2022-01-25:
@raju-apple Care to update the zapt file? If not, someone can take a look!

bzbarsky-apple, 2022-01-26:
So it looks like the chip-tool version of this has a timeout for an entire .yaml file (currently set at 90s), while the Darwin version has a per-step timeout (3s).

We can certainly bump the Darwin timeout.  What value would be useful there?

We can also add a way to specify a longer-than-default timeout for a given test step directly in the yaml, possibly in combination with also bumping the default to cover more cases.

@raju-apple Preferences?

raju-apple, 2022-01-26:
@bzbarsky-apple I like the second option , yeah if the yaml has a timeout specified in the step we should pick that up and if not it can fall back to the default. Maybe make the default as 5s or do you think its too much of a bump  ?

bzbarsky-apple, 2022-01-26:
5s default is easy.  Let me do this.  :)

closed 2022-01-27 by woody-apple