project-chip/connectedhomeip #14269: chip-tool wrong interpretation of device list in the description cluster, device-list attribute

chrisbeach59, 2022-01-25:
#### Problem
DUT reports type 22, rev 1, but chip tool interprets it as Type 1, Rev 0
[1643145748.328185][30126:30137] CHIP:DMG: ReportDataMessage =
[1643145748.328242][30126:30137] CHIP:DMG: {
[1643145748.328289][30126:30137] CHIP:DMG: 	AttributeReportIBs =
[1643145748.328354][30126:30137] CHIP:DMG: 	[
[1643145748.328405][30126:30137] CHIP:DMG: 		AttributeReportIB =
[1643145748.328473][30126:30137] CHIP:DMG: 		{
[1643145748.328534][30126:30137] CHIP:DMG: 			AttributeDataIB =
[1643145748.328598][30126:30137] CHIP:DMG: 			{
[1643145748.328661][30126:30137] CHIP:DMG: 				DataVersion = 0x0,
[1643145748.328723][30126:30137] CHIP:DMG: 				AttributePathIB =
[1643145748.328793][30126:30137] CHIP:DMG: 				{
[1643145748.328862][30126:30137] CHIP:DMG: 					Endpoint = 0x0,
[1643145748.328997][30126:30137] CHIP:DMG: 					Cluster = 0x1d,
[1643145748.329084][30126:30137] CHIP:DMG: 					Attribute = 0x0000_0000,
[1643145748.329161][30126:30137] CHIP:DMG: 				}
[1643145748.329238][30126:30137] CHIP:DMG: 					
[1643145748.329308][30126:30137] CHIP:DMG: 					Data = [
[1643145748.329376][30126:30137] CHIP:DMG: 						
[1643145748.329443][30126:30137] CHIP:DMG: 					],
[1643145748.329508][30126:30137] CHIP:DMG: 			},
[1643145748.329577][30126:30137] CHIP:DMG: 			
[1643145748.329630][30126:30137] CHIP:DMG: 		},
[1643145748.329715][30126:30137] CHIP:DMG: 		
[1643145748.329767][30126:30137] CHIP:DMG: 		AttributeReportIB =
[1643145748.329842][30126:30137] CHIP:DMG: 		{
[1643145748.329897][30126:30137] CHIP:DMG: 			AttributeDataIB =
[1643145748.330070][30126:30137] CHIP:DMG: 			{
[1643145748.330137][30126:30137] CHIP:DMG: 				DataVersion = 0x0,
[1643145748.330325][30126:30137] CHIP:DMG: 				AttributePathIB =
[1643145748.330620][30126:30137] CHIP:DMG: 				{
[1643145748.330708][30126:30137] CHIP:DMG: 					Endpoint = 0x0,
[1643145748.330787][30126:30137] CHIP:DMG: 					Cluster = 0x1d,
[1643145748.330863][30126:30137] CHIP:DMG: 					Attribute = 0x0000_0000,
[1643145748.330931][30126:30137] CHIP:DMG: 					ListIndex = Null,
[1643145748.331058][30126:30137] CHIP:DMG: 				}
[1643145748.331176][30126:30137] CHIP:DMG: 					
[1643145748.331241][30126:30137] CHIP:DMG: 					Data = 
[1643145748.331348][30126:30137] CHIP:DMG: 					{
[1643145748.331423][30126:30137] CHIP:DMG: 						0x0 = 22, 
[1643145748.331779][30126:30137] CHIP:DMG: 						0x1 = 1, 
[1643145748.332248][30126:30137] CHIP:DMG: 					},
[1643145748.332437][30126:30137] CHIP:DMG: 			},
[1643145748.332761][30126:30137] CHIP:DMG: 			
[1643145748.332830][30126:30137] CHIP:DMG: 		},
[1643145748.332909][30126:30137] CHIP:DMG: 		
[1643145748.332964][30126:30137] CHIP:DMG: 	],
[1643145748.333050][30126:30137] CHIP:DMG: 	
[1643145748.333109][30126:30137] CHIP:DMG: 	SuppressResponse = true, 
[1643145748.333161][30126:30137] CHIP:DMG: }
[1643145748.333691][30126:30137] CHIP:TOO: Descriptor.DeviceList response: 1 entries
[1643145748.333798][30126:30137] CHIP:TOO:   [1]: {
[1643145748.333858][30126:30137] CHIP:TOO:     Type: 1
[1643145748.333906][30126:30137] CHIP:TOO:     Revision: 0
[1643145748.333953][30126:30137] CHIP:TOO:   }

 #### Proposed Solution
Fix

chrisbeach59, 2022-01-25:
Interesting observation.  My observation is for a cpu platform Ameba. 
Another company has reported that they are seeing Device ID encoded in ID 1 and Version in ID 2.  They are using the nordic platform.  
Spec seems to indicate the it should be ID 0 and ID 1.

chrisbeach59, 2022-01-25:
The Te 7.5 branch 628d7ba19291a6ccc3bdaed693dabb1af98b0266 from RPI4 to RPI4 has ID1 and ID2 used. 
If I build from more modern cf1b25478bbcb93164b09f9546865430a3b7d8e7 commit the standalone app Is using ID0 and ID1 as I beleive is per spec.  But the chip-tool from the same commit also on linux exhibits the problem as it is still expecitng the data on ID1 and 2 not ID0 and ID1.

bzbarsky-apple, 2022-01-27:
The ids being off by 1 should have been fixed by #13731 (which is not included in https://github.com/project-chip/connectedhomeip/commit/628d7ba19291a6ccc3bdaed693dabb1af98b0266 but is included in https://github.com/project-chip/connectedhomeip/commit/cf1b25478bbcb93164b09f9546865430a3b7d8e7).

I just build chip-tool from https://github.com/project-chip/connectedhomeip/commit/cf1b25478bbcb93164b09f9546865430a3b7d8e7 and what I am seeing is:
```
[1643298461664] [83963:3317368] CHIP: [DMG] 					Data = 
[1643298461664] [83963:3317368] CHIP: [DMG] 					{
[1643298461664] [83963:3317368] CHIP: [DMG] 						0x0 = 22, 
[1643298461664] [83963:3317368] CHIP: [DMG] 						0x1 = 1, 
[1643298461664] [83963:3317368] CHIP: [DMG] 					},
....
[1643298461664] [83963:3317368] CHIP: [TOO] Descriptor.DeviceList response: 1 entries
[1643298461665] [83963:3317368] CHIP: [TOO]   [1]: {
[1643298461665] [83963:3317368] CHIP: [TOO]     Type: 22
[1643298461665] [83963:3317368] CHIP: [TOO]     Revision: 1
[1643298461665] [83963:3317368] CHIP: [TOO]   }
```
which looks correct.

@chrisbeach59 could you double-check what you are seeing with that SHA?

chrisbeach59, 2022-01-27:
Will do.  Need to finish up my TE7.5 testing.  Should be done today or tomorrow.  And will re-test.

chrisbeach59, 2022-03-03:
@bzbarsky-apple I just got to test this against TE8/rc3 and the behavior is correct. 
Closing the issue.

closed 2022-03-03 by chrisbeach59