project-chip/connectedhomeip #14275: chip app segfaults while trying to run Test_TC_DM_1_3_Simulated test case

wahajsyed, 2022-01-25:
#### Problem
- chip app segfaults while trying to run a simulated device test case

#### Config:
- Raspi 4
- Ubuntu 21.04 & 21.10 both
- All listed SDK and ZAP dependencies installed
- Build commands used:
```bash
# Chip App
scripts/examples/gn_build_test_example.sh app1

# Chip Tool
chip-tool: gn gen out/debug and ninja -C out/debug in chip-tool folder
```

#### Steps to Repro:
- Build chip-tool and chip app using the build instructions above
- Run the chip app using: 
```
 ./chip-app1 --command Test_TC_DM_1_3_Simulated --secured-device-port 5540
```
- Run the chip-tool using:
```
./chip-tool pairing onnetwork-long 0x654321 20202021 3842
```

<details>
<summary>Logs:</summary>

```
ubuntu@ubuntu:~/chip-certification-tool/backend/managed-docker-images/chip-tool$ ./chip-app1 --command Test_TC_DM_1_3_Simulated --secured-device-port 5540
[1643148897.546908][2805:2805] CHIP:DL: writing settings to file (/tmp/chip_factory.ini-BqXFun)
[1643148897.547217][2805:2805] CHIP:DL: renamed tmp file to file (/tmp/chip_factory.ini)
[1643148897.547611][2805:2805] CHIP:DL: writing settings to file (/tmp/chip_config.ini-y1PJv9)
[1643148897.547809][2805:2805] CHIP:DL: renamed tmp file to file (/tmp/chip_config.ini)
[1643148897.548067][2805:2805] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-pvAYZF)
[1643148897.548256][2805:2805] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1643148897.548516][2805:2805] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-lqPVsN)
[1643148897.549202][2805:2805] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1643148897.549271][2805:2805] CHIP:DL: NVS set: chip-counters/reboot-count = 1 (0x1)
[1643148897.549457][2805:2805] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-VAVBbm)
[1643148897.552623][2805:2805] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1643148897.552686][2805:2805] CHIP:DL: NVS set: chip-counters/total-operational-hours = 0 (0x0)
[1643148897.552854][2805:2805] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-wI5nEQ)
[1643148897.553369][2805:2805] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1643148897.553423][2805:2805] CHIP:DL: NVS set: chip-counters/boot-reason = 0 (0x0)
[1643148897.553564][2805:2805] CHIP:DL: writing settings to file (/tmp/chip_config.ini-Bh81l9)
[1643148897.554037][2805:2805] CHIP:DL: renamed tmp file to file (/tmp/chip_config.ini)
[1643148897.554090][2805:2805] CHIP:DL: NVS set: chip-config/regulatory-location = 0 (0x0)
[1643148897.554231][2805:2805] CHIP:DL: writing settings to file (/tmp/chip_config.ini-mJ9QI3)
[1643148897.554709][2805:2805] CHIP:DL: renamed tmp file to file (/tmp/chip_config.ini)
[1643148897.554764][2805:2805] CHIP:DL: NVS set: chip-config/location-capability = 0 (0x0)
[1643148897.554907][2805:2805] CHIP:DL: writing settings to file (/tmp/chip_config.ini-WTrBPt)
[1643148897.556127][2805:2805] CHIP:DL: renamed tmp file to file (/tmp/chip_config.ini)
[1643148897.556179][2805:2805] CHIP:DL: NVS set: chip-config/active-locale = "en-US"
[1643148897.556894][2805:2805] CHIP:DL: Got Ethernet interface: eth0
[1643148897.557398][2805:2805] CHIP:DL: Found the primary Ethernet interface:eth0
[1643148897.557945][2805:2805] CHIP:DL: Got WiFi interface: wlan0
[1643148897.558029][2805:2805] CHIP:DL: Failed to reset WiFi statistic counts
[1643148897.558208][2805:2805] CHIP:DL: Device Configuration:
[1643148897.558243][2805:2805] CHIP:DL:   Serial Number: TEST_SN
[1643148897.558270][2805:2805] CHIP:DL:   Vendor Id: 9050 (0x235A)
[1643148897.558295][2805:2805] CHIP:DL:   Product Id: 65279 (0xFEFF)
[1643148897.558323][2805:2805] CHIP:DL:   Hardware Version: 0
[1643148897.558351][2805:2805] CHIP:DL:   Setup Pin Code: 20202021
[1643148897.558379][2805:2805] CHIP:DL:   Setup Discriminator: 3842 (0xF02)
[1643148897.558408][2805:2805] CHIP:DL:   Manufacturing Date: (not set)
[1643148897.558434][2805:2805] CHIP:DL:   Device Type: 65535 (0xFFFF)
[1643148897.558829][2805:2805] CHIP:SVR: SetupQRCode: [MT:YNJV7FS212KA0648G00]
[... 1609 lines cut ...]
#4  0x0000aaaaaab6b440 in chip::DeviceLayer::Internal::GenericPlatformManagerImpl_POSIX<chip::DeviceLayer::PlatformManagerImpl>::_RunEventLoop()
    (this=0xaaaaaac48c28 <chip::DeviceLayer::PlatformManagerImpl::sInstance+16>) at ../../examples/placeholder/linux/third_party/connectedhomeip/src/include/platform/internal/GenericPlatformManagerImpl_POSIX.cpp:183
#5  0x0000aaaaaaaf3ee8 in chip::DeviceLayer::PlatformManager::RunEventLoop() (this=0xaaaaaac48c18 <chip::DeviceLayer::PlatformManagerImpl::sInstance>)
    at ../../examples/placeholder/linux/third_party/connectedhomeip/src/include/platform/PlatformManager.h:348
#6  0x0000aaaaaaaf43e4 in ChipLinuxAppMainLoop() () at ../../examples/placeholder/linux/third_party/connectedhomeip/examples/platform/linux/AppMain.cpp:434
#7  0x0000aaaaaaac31b0 in main(int, char**) (argc=5, argv=0xfffffffff388) at ../../examples/placeholder/linux/main.cpp:34
(gdb) 
```

The following diff fixes the issue for now:
```
diff --git a/examples/placeholder/linux/include/TestCommand.h b/examples/placeholder/linux/include/TestCommand.h
index 40e111fc3cd..232e9005e8f 100644
--- a/examples/placeholder/linux/include/TestCommand.h
+++ b/examples/placeholder/linux/include/TestCommand.h
@@ -82,7 +82,6 @@ public:
             TestCommand * command = reinterpret_cast<TestCommand *>(arg);
             command->isRunning    = true;
             command->NextTest();
-            chip::DeviceLayer::PlatformMgr().RemoveEventHandler(OnPlatformEvent, arg);
             break;
         }
     }
 ```
</details>

bzbarsky-apple, 2022-01-26:
@vivien-apple

vivien-apple, 2022-01-26:
As discussed, `ScheduleWork` should do the job.

woody-apple, 2022-02-09:
Isn't this fixed now? Please re-open if not!

closed 2022-02-09 by woody-apple