project-chip/connectedhomeip #14278: Bootstrap command failed setting up Python environment

nanobert, 2022-01-25:
#### Problem
 * Bootstrap command failed setting up Python environment (pigweed)
 * Error: `FileNotFoundError: [Errno 2] No such file or directory: 'gn'`

#### System configuration:
Raspberry PI 4 with ubuntu server 32bit
ubuntu@ubuntu:~/share32/connectedhomeip/scripts$ cat /etc/os-release
NAME="Ubuntu"
VERSION="20.04.3 LTS (Focal Fossa)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 20.04.3 LTS"
VERSION_ID="20.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=focal
UBUNTU_CODENAME=focal

ubuntu@ubuntu:~/share32/connectedhomeip/scripts$ uname -r
5.4.0-1050-raspi

#### Versions:
```
ubuntu@ubuntu:~/share32$ python3 --version
Python 3.8.10
ubuntu@ubuntu:~/share32$ python --version
Python 3.9.5
```

#### Steps:
`ubuntu@ubuntu:~/share32$ git clone https://github.com/project-chip/connectedhomeip.git`
`ubuntu@ubuntu:~/share32/connectedhomeip$ git submodule update --init --recursive`
`ubuntu@ubuntu:~/share32/connectedhomeip$ source scripts/activate.sh`

#### Logs:
```
ubuntu@ubuntu:~/share32/connectedhomeip$ source scripts/activate.sh

  WELCOME TO...

           ░▓░
           ▓█▓
           ▓█▓                                                     ▒█     ▒█
      ▒██▒▒▓██▒███▒                ░▒▓▒░  ░▒▓▒░       ░░▓█▒░ ░█  █████████████░    ░▒█▒░       ░░▒░
       ░▓█████▓██░               ▒█▒░░▒▓██▓▒░░▒█▒   ░█▓▒░░▒████    █▓░    █▓░    ▒█▒░░░▒█▓░   ██▒░░
    ▒█▒░         ░██▒           ░█░     █▓     ░█░ ░▓▒      ░▓█    █▓     █▓    ▒█░░    ░██  ▒▓
    ░▓██▓░     ░██▓█░           ░█      ▓█      █░ ░█░       ██    █▓     █▓    ▓██████████  ▒█
     ░▓███▒   ▒███▒░            ░█      ▓█      █░  ██░    ░███    █▓     █▓    ░█▒░         ▒█
 ░▒████████░ ░███▓▓█▓▓▒         ░█      ▒▒      █░   ░█▓██▓█░▒█    ░▓▓█░  ░▓▓█░   ▒▓▓██▓█░   ▒▓
 ░██▒░  ▒██▒ ▒██░  ░▒█▓
         ░▓░ ░▓░

  BOOTSTRAP! Bootstrap may take a few minutes; please be patient.

Downloading and installing packages into local source directory:

  Setting up CIPD package manager...done (4.7s)
  Setting up Python environment.....Traceback (most recent call last):
  File "/home/ubuntu/share32/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 715, in <module>
    sys.exit(main())
  File "/home/ubuntu/share32/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 707, in main
    return EnvSetup(**vars(parse())).setup()
  File "/home/ubuntu/share32/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 419, in setup
    result = step(spin)
  File "/home/ubuntu/share32/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 538, in virtualenv
    if not virtualenv_setup.install(
  File "/home/ubuntu/share32/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 301, in install
    install_packages(gn_target)
  File "/home/ubuntu/share32/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 269, in install_packages
    subprocess.check_call(gn_cmd,
  File "/usr/lib/python3.8/subprocess.py", line 359, in check_call
    retcode = call(*popenargs, **kwargs)
  File "/usr/lib/python3.8/subprocess.py", line 340, in call
    with Popen(*popenargs, **kwargs) as p:
  File "/usr/lib/python3.8/subprocess.py", line 858, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.8/subprocess.py", line 1704, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'gn'
ubuntu@ubuntu:~/share32/connectedhomeip$ 
```

woody-apple, 2022-01-26:
@nanobert Have you followed all the pre-reqs from here? 

https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/BUILDING.md

Looks like gn isn't dropping in for some reason from CIPD?

nanobert, 2022-01-31:
I followed the setup, but note: I used a 32bit image.
There is no gn available for the target and it needed be be compiled from scratch.
After this, the next issue appeared; pigweed. I installed pigweed from pip.
Then protobuf woudn't install and I stopped

closed 2022-01-31 by nanobert

bpetersen-grundfos, 2022-02-05:
Had the same issue, changed to PI OS Lite 64bit then it worked

sihan6, 2022-02-09:
I have the same issue, and I need to use 32 bit system. Have you solved the problem?