project-chip/connectedhomeip #14304: Address review comment on PR 13742

bzbarsky-apple, 2022-01-26:
#### Problem
It doesn't look like my review comment on #13724 was addressed in any sort of reasonable way.

Again:

1. We are calling a function that copies data but if the destination is too small it will return error and not copy anything.
2. We are not checking whether it failed or not.
3. We are using the buffer it was copying to, which might have random data if no copy happened.

If something ensures that our data to copy is small enough to fit, what that "something" is should be documented.

I did see that some checks against "8" and "32" were added.  Where are those numbers coming from?  If they're meant to match buffer sizes, why are we not comparing to the sizes of the buffers instead?  What happens if the buffer sizes change and the checks get out of sync?

 #### Proposed Solution
Address the comments.

@jadhavrohit924 @andy31415 @carol-apple

bzbarsky-apple, 2022-01-26:
In particular, this bit:
```
CopySpanToMutableSpan(update.updateToken, updateToken);
```

jadhavrohit924, 2022-01-27:
According to my understanding, 1) we need to check with actual size of buffer instead of 8 and 32 bytes.
2) we need to insure that copy happened or not.

but we are copying only if size of buffer is big enough otherwise we are returning error

bzbarsky-apple, 2022-01-27:
If the check for size guarantees the call can't fail, that's fine.  We just need to document where the checks are, and make sure the checks are checking the right quantities.

turon, 2022-01-31:
Needed for v1. Removing `v1_triage_split_4`.

dhrishi, 2022-02-03:
@jadhavrohit924 Is this addressed? Can we close?

jadhavrohit924, 2022-02-04:
@dhrishi yes

closed 2022-02-04 by dhrishi