project-chip/connectedhomeip #14314: Is there a way to make this code less error-prone for consumers?

woody-apple, 2022-01-26:
The consumer doesn&#x27;t need to know the exact OID value.

---

https://github.com/project-chip/connectedhomeip/blob/20c1f60a289e293246020a13aa85f81de5fec8c8/src/controller/ExampleOperationalCredentialsIssuer.cpp#L99-L104

---

###### This issue was generated by [todo-issue](https://github.com/DerJuulsn/todo-issue) based on a `TODO` comment in 20c1f60a289e293246020a13aa85f81de5fec8c8. It&#x27;s been assigned to @emargolis because they committed the code.

woody-apple, 2022-02-09:
Agree, let's move that out.

emargolis, 2022-04-11:
This was partially addressed in #17190 

There still TODO comment from @bzbarsky-apple that should be addressed:
```
So... This is fine as far as it goes, but the actual TODO was about seeing whether here was a way to make the "link up the NOC and ICA cert and RCA cert just right" code that's more or less duplicated across all the credential providers harder to mess up.

For example, maybe we should have a utility for "create an operational root cert" that takes the validity start, validity end, issuer id, fabric id, and key and does the setup of the DN, the listing of the DN twice in the X509CertRequestParams, etc. And then returns some piece of data that we can pass to the "create an ICA cert" or "create a NOC" bits that will automatically set the right "parent" DN, etc....
```