project-chip/connectedhomeip #14318: third_party/openthread/ot-ifx: commit hash not on default branch cyw30739

tewarid, 2022-01-26:
third_party/openthread/ot-ifx does not contain commit has da522560fbff49f6b27bfef2139e898018c5f0f4 on default branch cyw30739
https://github.com/project-chip/connectedhomeip/blob/20c1f60a289e293246020a13aa85f81de5fec8c8/.gitmodules#L85

turon, 2022-01-31:
This is blocking my local build currently after reset to `origin/master`.

Removing `v1_triage_split_10` tag.

turon, 2022-01-31:
Try `git submodule update --init -f --recursive`.

turon, 2022-01-31:
May have been introduced in PR #13491.

Entered tree as commit 1af90eb

Temp work-around to revert to dd209ce

turon, 2022-02-08:
@woody-apple Are other Mac devs not seeing this?

turon, 2022-02-08:
I pretty much just had to reclone the entire repo from scratch.
No combination of commands would restore my local tree:

```
git reset --hard origin/master
git restore third_party/openthread/ot-ifx
git submodule update --init --recursive -f
...
```

woody-apple, 2022-02-09:
> @woody-apple Are other Mac devs not seeing this?

Not that I know of? However, it sounds like this is pretty broken.

ItoddchouFX, 2022-02-09:
@turon  @tewarid  The issue is corrected in Infineon/ot-ifx-release. Let me if it works.

tewarid, 2022-02-09:
Thanks @ItoddchouFX - our yocto linux build works now.

closed 2022-02-09 by tewarid