project-chip/connectedhomeip #14321: OTA: End device (Requestor) crashes if it cannot reach the Provider

mykrupp, 2022-01-26:
#### Problem

Expected behavior: announce-ota-provider should error out when no Provider node ID has been registered. One reason that the Provider node ID would not be registered is if the provider has not been commissioned over thread

When the end device receives the Announce OTA Provider command and the Provider node ID indicated in that command is not reachable, the end device crashes. The function HandleNodeIdResolve in Discovery_ImplPlatform.cpp is passed a DnssdService pointer that contains incomplete information. Attributes mKey, mData contain null fields and mDataSize seems uninitialized.

Steps to reproduce: 
Follow - [https://github.com/selissia/connectedhomeip/edit/master/examples/ota-requestor-app/efr32/README.md](url)
But do not commission provider

Call stack in Ozone: 
[image]

Hardfault register details:
[image]

 #### Proposed Solution

Either a null check around the provider node ID fields, or an error code indicating that the provider has not been comissioned.

holbrookt, 2022-01-31:
@selissia should this issue be assigned to you as it's efr32? If not, can you help assign?

selissia, 2022-01-31:
Yes, this is a DNS issue not specific to EFR32. As such I initially assigned it to @vivien-apple but @woody-apple unassigned it, I'm not sure why.

holbrookt, 2022-01-31:
Maybe an automated script? Feel free to reassign if you think that's right